MarlinFirmware/Marlin #23095: [BUG] X and Y axis remain trusted after a M84, only Z axis is marked as untrusted

mriscoc, 2021-11-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Tested in Ender3v2 with the last bugfix:

Do an auto home (G29) and wait to finish and disable steppers with M84. The system indicates X and Y axis as trusted, and Z as untrusted.
 

### Expected behavior

All axes will be marked as untrusted after an M84

### Actual behavior

only Z axis will be marked as untrusted after an M84

### Steps to Reproduce

Do an auto home (G29) and wait to finish and disable steppers with M84. The system indicates X and Y axis as trusted, and Z as untrusted.

### Version of Marlin Firmware

Latest bugfix 2.0.x

### Printer model

Creality Ender 3v2

### Electronics

STM32F103RET6

### Add-ons

none

mriscoc, 2022-01-08:
was this bug fixed?

thisiskeithb, 2022-01-08:
> was this bug fixed?

Your PR is still pending, so probably not.

zinen, 2022-03-11:
Testing this on Ender5 Pro gives the same issue.

Martian-35, 2023-03-24:
The Z axis remails trusted now too. If the gentry is lowered manually and then you run a probe cycle, it crashes.

Farva42, 2023-03-24:
I started looking at this. When m84 is called it will only try and set the Z axis to disabled but all axis will be disabled regardless of the flag not being set. If you call G28 after X&Y will still show trusted with Z showing untrusted but it will not do a z raise before attempting to home. 

If you do two M84's in a row the second one will correctly set all three axis to disabled and set all three to untrusted but a G28 after will still act like all axis are trusted and neglect to do a Z raise.

classicrocker883, 2024-04-28:
this issue is still present

closed 2026-10-15 by thinkyhead