MarlinFirmware/Marlin #23098: [BUG] New confusing compiler warning regarding Sensorless Homing

bill-orange, 2021-11-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin compiles with a new warning (well intended)which is confusing in my configuration.  Here's the warning:

> Marlin\src\inc\Warnings.cpp:544:4: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled." [-Wcpp]
>   544 |   #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled."

I am using an SKR1,3 with TMC2130 drivers.  There is a diag pin on this combination of driver/board but no jumper.  I believe the 2.0 board has a jumper.

I am not using sensorless homing.  Should I, (a) do nothing or (b) cut the diag PIN (not jumper)

### Bug Timeline

First noticed 11/9/2021

### Expected behavior

No warning or perhaps a reworded warning.

### Actual behavior

Confusing warning that does not apply well to at least one common hardware/ software configuration,  TMC2130, SKR1.3, No sensorless homing.

### Steps to Reproduce

a. Recompile
b. Observe warning

### Version of Marlin Firmware

current bugfix version.  Can not run M115 at present

### Printer model

Anet A8 with SKR1.3, TMC2130 drivers

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

I guess my post is a two part question.  On my present hardware software configuration do I need to change anything to not use sensorless homing?

Should the warning be rewarded a bit.  Since I don't know the answer to my question I am not sure what the new wording should be.  Perhaps: 

Motherboard DIAG jumpers must be removed (if present) when SENSORLESS_HOMING is disabled

or possibly depending on the answer to my question:

Motherboard DIAG jumpers ( or DIAG pin) must be removed  when SENSORLESS_HOMING is disabled

thisiskeithb, 2021-11-09:
That warning was added to alert users of the required change to boards/drivers since it was an extremely common troubleshooting issues.

>  There is a diag pin on this combination of driver/board but no jumper.

The SKR 1.3 uses DIAG jumpers. They are below E1 and `Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled."` ...or must be removed if you are not using sensorless homing. [From the manual](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/SKR%20V1.3-Instructions.pdf): 

[image]

closed 2021-11-09 by thisiskeithb

bill-orange, 2021-11-09:
I did not realize those were diag jumpers.  I expected diag jumpers to be under the driver plugs.  Sorry about that.

I think that there still may be need for clarification of the warning.  Most people using TMC2130 are likely using them in SPI mode, as am I.  As the illustration in your reply shows, in SPI mode all the 'purple' contacts shown with jumpers are to be  jumpered.  I believe this gives software control of sensorless homing vs the usual homing along with software control of microstepping.  

I am not running sensorless homing and I have never removed those jumpers.

thisiskeithb, 2021-11-09:
> I am not running sensorless homing and I have never removed those jumpers.

If your drivers do not have DIAG pins soldered in place, then that would achieve the same thing.

bill-orange, 2021-11-09:
Ah, but they are in place. In SPI mode, control over all of this jumper ‘stuff’ is done via software. 

At least that’s my understanding. In SPI mode the jumpers are needed.