MarlinFirmware/Marlin #23111: [BUG] Babystepping moves only to a single direction, based on direction machine was moving previously

Justase, 2021-11-11:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Z axis Babystesps are moving to a single direction disregarding how I want to get it moving. If the machine was moving manually or based on the G-code up, then it will continue moving up no matter which direction the knob was turned. The same happens to down position. It looks like babysteps are not controlling direction pin for the stepper drivers.
Drivers that I use are MKS servo42b, configured as default Allegro drivers.
Marlin version is 2.0.9.2, but the same was happening on 2.0.7.2. 
The board now is BTT Octopus, previously BTT SKR 1.4
Tried different settings on configuration files, but I'm out of ideas how to make it working as it should.

### Expected behavior

To move up, when asked to move up.
To move down, when asked to move down.

### Actual behavior

If previously was moving up (manual control or g-code) then it will move up no matter how I direct.
If previously was moving down (manual control or g-code) then it will move down no matter how I direct.

### Version of Marlin Firmware

2.0.9.2

### Printer model

SecKit SK-Go2

### Electronics

BTT Octopus

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7518766/Marlin.zip)

grawi-sikorsky, 2021-11-11:
I can confirm that something is wrong with babystepping. In my case - corexy with bed endstop on Z_MAX pin and piezo probe on Z_MIN pin, babystepping is working only in "negative" direction (Babystep::steps negative values). That means bed can go up, but when trying to add babystepp in positive direction (bed should go down) Marlin just crashes and resets itself.

Mainboard SKR 2.0 with LV8729 drivers, AUTO_BED_LEVELING_BILINEAR, NOZZLE_AS_PROBE, latest bugfix-2.0.x

Justase, 2021-11-11:
I think I found the problem. Ended up adding extra delays in the stepper.cpp file.
Not the best solution for upgradability, but it works, and delays are very selective for very specific part of code, not global. 
I was surprised on how much time had to be added for steppers to take into account the change of direction.

```
+     //JE 20211111 - Added extra delays for applying DIR                                             
    #define BABYSTEP_AXIS(AXIS, INV, DIR) do{           \
      const uint8_t old_dir = _READ_DIR(AXIS);          \
      _ENABLE_AXIS(AXIS);                               \
      DIR_WAIT_BEFORE();                                \
      _APPLY_DIR(AXIS, _INVERT_DIR(AXIS)^DIR^INV);      \
+      DELAY_NS(1500);                                   \
      DIR_WAIT_AFTER();                                 \
      _SAVE_START();                                    \
      _APPLY_STEP(AXIS, !_INVERT_STEP_PIN(AXIS), true); \
      _PULSE_WAIT();                                    \
      _APPLY_STEP(AXIS, _INVERT_STEP_PIN(AXIS), true);  \
      EXTRA_DIR_WAIT_BEFORE();                          \
      _APPLY_DIR(AXIS, old_dir);                        \
      EXTRA_DIR_WAIT_AFTER();                           \
+      DELAY_NS(1000);                                   \ 
    }while(0)
```

borland1, 2021-11-14:
More appropriate, might be to add those delays in `ConfigurationAdv.h` under: 

```
2045 // @section extras,  

2112 //#define MINIMUM_STEPPER_POST_DIR_DELAY 650 + 1000
2113 //#define MINIMUM_STEPPER_PRE_DIR_DELAY 650 + 1500
```

Justase, 2021-11-16:
I wanted to set delay only to baby stepping in stepper.cpp without affecting other movements, however I guess it may be worth to have it set globally as @borland1 suggests. I have a feeling that during normal operation steppers may run several steps wrong direction before taking into account change in direction. On the other hand, prints had no noticeable defects

borland1, 2021-11-16:
Those delays are only nanosecond scale, so might not make much difference globally.

It appear from the associated commented lines in `Configuration_adv.h` that the suggested delay values (in nanoseconds) were taken from the manufacturer's datasheet of each stepper driver model, but I wonder why those defined values are commented out by default?

Justase, 2021-11-16:
Looking other way round In my case at 100mm/s machine is doing step every 100us, so 1.5us delay for DIR change to be taken on servo42b board does not look significant even without taking into account accelerations. I'll try to set delays globally. 

In terms of commented out delays on `Configuration_adv.h`, there are some default definitions per driver type on `Conditionals_adv.h` starting at line 820.
```
#ifndef MINIMUM_STEPPER_POST_DIR_DELAY
  #if HAS_DRIVER(TB6560)
    #define MINIMUM_STEPPER_POST_DIR_DELAY 15000
  #elif HAS_DRIVER(TB6600)
    #define MINIMUM_STEPPER_POST_DIR_DELAY 1500
  #elif HAS_DRIVER(DRV8825)
    #define MINIMUM_STEPPER_POST_DIR_DELAY 650
  #elif HAS_DRIVER(LV8729)
    #define MINIMUM_STEPPER_POST_DIR_DELAY 500
  #elif HAS_DRIVER(A5984)
    #define MINIMUM_STEPPER_POST_DIR_DELAY 400
  #elif HAS_DRIVER(A4988)
    #define MINIMUM_STEPPER_POST_DIR_DELAY 200
  #elif HAS_TRINAMIC_CONFIG || HAS_TRINAMIC_STANDALONE
    #define MINIMUM_STEPPER_POST_DIR_DELAY 60
  #else
    #define MINIMUM_STEPPER_POST_DIR_DELAY 0   // Expect at least 10µS since one Stepper ISR must transpire
  #endif
#endif
```

borland1, 2021-11-16:
Thanks for pointing that out in `Conditional_adv.h`.   As such, in my printer's config case, where I have TRINAMIC drivers on XYZ and DRV8825 driver on E0, the `MINIMUM_STEPPER_POST_DIR_DELAY` (and I assume `PRE` too) for all drivers is set to worst case of 650ns.

Tannoo, 2021-11-22:
This can be closed now?

closed 2021-11-22 by Justase