MarlinFirmware/Marlin #23114: Can't compile error: token "{" is not valid in preprocessor expressions

sergiofagundes, 2021-11-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Using platformio, compilation of Marlin 2.0.9.2 (bugfix also) ends of the following message:

Compiling .pio\build\mks_robin_e3\src\src\lcd\marlinui.cpp.o
In file included from Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.h:28,
                 from Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:34:
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:93:30: error: token "{" is not valid in preprocessor expressions
   93 |   #define DELAY_CYCLES(X) do { SmartDelay<IS_CONSTEXPR(X), IS_CONSTEXPR(X) ? X : 0> _smrtdly_X(X); } while(0)
      |                              ^
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:216:29: note: in expansion of macro 'DELAY_CYCLES'
  216 | #define DELAY_NS(x)         DELAY_CYCLES(_NS_TO_CYCLES(x))
      |                             ^~~~~~~~~~~~
Marlin\src\lcd\dogm\../../inc/../pins/stm32f1/pins_MKS_ROBIN_E3_common.h:221:36: note: in expansion of macro 'DELAY_NS'
  221 |   #define BOARD_ST7920_DELAY_1     DELAY_NS(125)
      |                                    ^~~~~~~~
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:49:29: note: in expansion of macro 'BOARD_ST7920_DELAY_1'
   49 |   #if LCD_ST7920_DELAY_1 || BOARD_ST7920_DELAY_1 || CPU_ST7920_DELAY_1
      |                             ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:93:30: error: token "{" is not valid in preprocessor expressions
   93 |   #define DELAY_CYCLES(X) do { SmartDelay<IS_CONSTEXPR(X), IS_CONSTEXPR(X) ? X : 0> _smrtdly_X(X); } while(0)
      |                              ^
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:216:29: note: in expansion of macro 'DELAY_CYCLES'
  216 | #define DELAY_NS(x)         DELAY_CYCLES(_NS_TO_CYCLES(x))
      |                             ^~~~~~~~~~~~
Marlin\src\lcd\dogm\../../inc/../pins/stm32f1/pins_MKS_ROBIN_E3_common.h:224:36: note: in expansion of macro 'DELAY_NS'
  224 |   #define BOARD_ST7920_DELAY_2     DELAY_NS(125)
      |                                    ^~~~~~~~
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:65:29: note: in expansion of macro 'BOARD_ST7920_DELAY_2'
   65 |   #if LCD_ST7920_DELAY_2 || BOARD_ST7920_DELAY_2 || CPU_ST7920_DELAY_2
      |                             ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:93:30: error: token "{" is not valid in preprocessor expressions
   93 |   #define DELAY_CYCLES(X) do { SmartDelay<IS_CONSTEXPR(X), IS_CONSTEXPR(X) ? X : 0> _smrtdly_X(X); } while(0)
      |                              ^
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:216:29: note: in expansion of macro 'DELAY_CYCLES'
  216 | #define DELAY_NS(x)         DELAY_CYCLES(_NS_TO_CYCLES(x))
      |                             ^~~~~~~~~~~~
  227 |   #define BOARD_ST7920_DELAY_3     DELAY_NS(125)
      |                                    ^~~~~~~~
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:81:29: note: in expansion of macro 'BOARD_ST7920_DELAY_3'
   81 |   #if LCD_ST7920_DELAY_3 || BOARD_ST7920_DELAY_3 || CPU_ST7920_DELAY_3
      |                             ^~~~~~~~~~~~~~~~~~~~
*** [.pio\build\mks_robin_e3\src\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp.o] Error 1
======================================================= [FAILED] Took 591.22 seconds =======================================================
Environment    Status    Duration
-------------  --------  ------------
mks_robin_e3   FAILED    00:09:51.220
================================================== 1 failed, 0 succeeded in 00:09:51.220 ==================================================

The version 2.0.9.1 pass compilation without any error.

### Bug Timeline

new

### Expected behavior

Complete compilation

### Actual behavior

Error: error: token "{" is not valid in preprocessor expressions

### Version of Marlin Firmware

2.0.9.2

### Printer model

Kossel Delta

### Electronics

Mks Robin E3D

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

[configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/7522886/configuration_files.zip)

ellensp, 2021-11-11:
Provided configs compiles without issue on marlin release 2.0.9.2

ellensp, 2021-11-11:
delete your Marlin/.pio folder and and the .platfomio directory in you user home directory and try again

sergiofagundes, 2021-11-11:
> delete your Marlin/.pio folder and and the .platfomio directory in you user home directory and try again

Tried deleting these folders, didn't work, same error messages.

Version 2.0.9.1 compiles without errors, but the 2.0.9.2 or the bugfix gives error compiling.

ellensp, 2021-11-11:
Please attach log of compile (In particular all the version information at the start)

sergiofagundes, 2021-11-12:
Following is the compile log:

Processing mks_robin_e3 (platform: ststm32@~12.1; board: genericSTM32F103RC; framework: arduino)
---------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RC (48k RAM. 256k Flash)        
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.3
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <U8glib-HAL> 0.5.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\mks_robin_e3\src\src\inc\Warnings.cpp.o
Compiling .pio\build\mks_robin_e3\src\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp.o
Compiling .pio\build\mks_robin_e3\src\src\lcd\menu\menu_bed_leveling.cpp.o
Compiling .pio\build\mks_robin_e3\src\src\lcd\menu\menu_configuration.cpp.o
In file included from Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.h:28,
                 from Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:34:
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:93:30: error: token "{" is not valid in preprocessor expressions  
   93 |   #define DELAY_CYCLES(X) do { SmartDelay<IS_CONSTEXPR(X), IS_CONSTEXPR(X) ? X : 0> _smrtdly_X(X); } while(0)
      |                              ^
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:216:29: note: in expansion of macro 'DELAY_CYCLES'
  216 | #define DELAY_NS(x)         DELAY_CYCLES(_NS_TO_CYCLES(x))
      |                             ^~~~~~~~~~~~
Marlin\src\lcd\dogm\../../inc/../pins/stm32f1/pins_MKS_ROBIN_E3_common.h:221:36: note: in expansion of macro 'DELAY_NS'
  221 |   #define BOARD_ST7920_DELAY_1     DELAY_NS(125)
      |                                    ^~~~~~~~
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:49:29: note: in expansion of macro 'BOARD_ST7920_DELAY_1'
   49 |   #if LCD_ST7920_DELAY_1 || BOARD_ST7920_DELAY_1 || CPU_ST7920_DELAY_1
      |                             ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:93:30: error: token "{" is not valid in preprocessor expressions  
   93 |   #define DELAY_CYCLES(X) do { SmartDelay<IS_CONSTEXPR(X), IS_CONSTEXPR(X) ? X : 0> _smrtdly_X(X); } while(0)
      |                              ^
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:216:29: note: in expansion of macro 'DELAY_CYCLES'
  216 | #define DELAY_NS(x)         DELAY_CYCLES(_NS_TO_CYCLES(x))
      |                             ^~~~~~~~~~~~
Marlin\src\lcd\dogm\../../inc/../pins/stm32f1/pins_MKS_ROBIN_E3_common.h:224:36: note: in expansion of macro 'DELAY_NS'
  224 |   #define BOARD_ST7920_DELAY_2     DELAY_NS(125)
      |                                    ^~~~~~~~
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:65:29: note: in expansion of macro 'BOARD_ST7920_DELAY_2'
   65 |   #if LCD_ST7920_DELAY_2 || BOARD_ST7920_DELAY_2 || CPU_ST7920_DELAY_2
      |                             ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:93:30: error: token "{" is not valid in preprocessor expressions  
   93 |   #define DELAY_CYCLES(X) do { SmartDelay<IS_CONSTEXPR(X), IS_CONSTEXPR(X) ? X : 0> _smrtdly_X(X); } while(0)
      |                              ^
Marlin\src\lcd\dogm\../../HAL/shared/Delay.h:216:29: note: in expansion of macro 'DELAY_CYCLES'
  216 | #define DELAY_NS(x)         DELAY_CYCLES(_NS_TO_CYCLES(x))
      |                             ^~~~~~~~~~~~
Marlin\src\lcd\dogm\../../inc/../pins/stm32f1/pins_MKS_ROBIN_E3_common.h:227:36: note: in expansion of macro 'DELAY_NS'
  227 |   #define BOARD_ST7920_DELAY_3     DELAY_NS(125)
      |                                    ^~~~~~~~
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:81:29: note: in expansion of macro 'BOARD_ST7920_DELAY_3'
   81 |   #if LCD_ST7920_DELAY_3 || BOARD_ST7920_DELAY_3 || CPU_ST7920_DELAY_3
      |                             ^~~~~~~~~~~~~~~~~~~~
*** [.pio\build\mks_robin_e3\src\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp.o] Error 1
========================================= [FAILED] Took 52.16 seconds =========================================
Environment    Status    Duration
-------------  --------  ------------
mks_robin_e3   FAILED    00:00:52.158
==================================== 1 failed, 0 succeeded in 00:00:52.158 ====================================

sergiofagundes, 2021-11-12:
So after looking your compilation log is just like mine (and your board too), it can only be the configurations settings causing this issue.

I download the again a firmware, but this time I didn't change anything in it, and voialá: it compiles.

Now I'm sure it's there's some setting in the  configuration and configuration_adv causing this. Time to hunt the culprid. 

Thanks for your help !

sergiofagundes, 2021-11-12:
Problem found:

This error happens if I turn on:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

But hey, I need this on, it's only the display I have. If using another display no error in compilation.

ellensp, 2021-11-12:
provided configs have REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled
and still work fine for me.

delete Marlin/.pio and zip up your entire marlin folder and attach it  (should be small enough if you download the zip, and not a git clone)

sergiofagundes, 2021-11-12:
> provided configs have REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled and still work fine for me.
> 
> delete Marlin/.pio and zip up your entire marlin folder and attach it (should be small enough if you download the zip, and not a git clone)

I have attached the entire marlin firmware (removing the /.pio folder).

[Marlin-2.0.9.2 ROBIN E3D.zip](https://github.com/MarlinFirmware/Marlin/files/7527084/Marlin-2.0.9.2.ROBIN.E3D.zip)

ellensp, 2021-11-12:
There are 162 changed files when compared with marlin release 2.0.9.2
You are running 2.0.x which is release 2.0.9.2 + bugfixes

ellensp, 2021-11-12:
Using Marlin 2.0.x I can get this issue.
If you enable the following in Configuration.h It compiles 
#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(250)

But it shouldn't fail in such an obscure way...
Looking into why...

ellensp, 2021-11-12:
The zip you provided has a modified Marlin/src/pins/stm32f1/pins_MKS_ROBIN_E3_common.h
This contains 
```
#ifndef BOARD_ST7920_DELAY_1
  #define BOARD_ST7920_DELAY_1     DELAY_NS(125)
#endif
#ifndef BOARD_ST7920_DELAY_2
  #define BOARD_ST7920_DELAY_2     DELAY_NS(125)
#endif
#ifndef BOARD_ST7920_DELAY_3
  #define BOARD_ST7920_DELAY_3     DELAY_NS(125)
#endif
```
Which is not the way it is done now
It should be 
```
#if ENABLED(U8GLIB_ST7920)
  #ifndef BOARD_ST7920_DELAY_1
    #define BOARD_ST7920_DELAY_1             125
  #endif
  #ifndef BOARD_ST7920_DELAY_2
    #define BOARD_ST7920_DELAY_2             125
  #endif
  #ifndef BOARD_ST7920_DELAY_3
    #define BOARD_ST7920_DELAY_3             125
  #endif
#endif

```

As it is in the current pins_MKS_ROBIN_E3_common.h for marlin 2.0.x
 https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_E3_common.h#L224

ellensp, 2021-11-12:
replace the modified file with the original.

closed 2021-11-12 by ellensp