MarlinFirmware/Marlin #23115: 3d printer keeps resetting itself when i try to print

peguri47, 2021-11-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

i few days ago i use my 3d printer for around an hour or so, i turned off and i went to bed... a few days later i tried to used again
and it stars the abl (does 2 points) and then moves a little ( like trying to go to the next one) and stops, the spitts out a msg in the tft "V83 Hardcore stored settings retrived (609 bytes; crc 9969) 

### Bug Timeline

as i sayd the machine was off of a few days, and then this appear 

### Expected behavior

the printer moves, around the probing points and then starts printing

### Actual behavior

does 2 points and stops trying to go to the next

### Steps to Reproduce

to be honest, idk 

### Version of Marlin Firmware

2.0.9

### Printer model

farily modified anet e10 

### Electronics

SKR V1.3 

### Add-ons

TFT35V2.0

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

![photo4925114903736461980](https://user-images.githubusercontent.com/94144989/141375155-94b0510b-a640-4084-ad69-5218be11cc50.jpg

https://user-images.githubusercontent.com/94144989/141375159-666308ff-4b07-4691-adcc-6bf5867069be.mp4

)
as i said no modifications had been mad before or after this issue, i already tried the newest marlin version, no changes
i tried reflashing the same one and to the tft, no changes
i tried M502, M500 and then M501, didnt change anything, im about to trash the hole board, so i hope that u guys have any ideas

futuretech208, 2021-11-14:
I had similar issue for 2 months on my printer. The solution was loose connections and mainly the faulty cheap chinese thermistor. I replaced the thermistor with 300 deg C rating and things are working smooth now. The glass thermistor with the PTFE tube was causing the problem. Also, change PTFE to fibreglass insulation.

Tannoo, 2021-11-16:
Yes, this is a wiring issue. The resetting is likely due to something shorting out. Check everything... not just thermistor wiring.

peguri47, 2021-11-17:
> Yes, this is a wiring issue. The resetting is likely due to something shorting out. Check everything... not just thermistor wiring.

i don't believe that it's that, now it does that, but never actually arrives at abl, like does the auto home, then go z safe homing, and after that nothing moves, the temp doesn't go up... someone told me that the board it's done 
but I'm gonna check

peguri47, 2021-11-18:
i agree, maybe it's a hardware issue, because i move up to the same point, check everything, and everything works

closed 2021-11-18 by thisiskeithb