MarlinFirmware/Marlin #23120: [FR] (Software servo combination)

kj4sst, 2021-11-12:
### Are you looking for hardware support?

I am building a custom printer using the BTT Octopus board.

### Describe the feature you want

I'm currently working on a tool head that uses a servo to switch between two extruders and a z endstop. In order for the servo to operate properly in the firmware the switching nozzle servo and z probe servo have to be defined as separate servos (having those two servos being defined to the same pin in the board config file) and deactivate_servos_after_move has to be defined. I'd like to have the servo enabled to ensure that it holds its position.
```
#define SWITCHING_NOZZLE
#if ENABLED(SWITCHING_NOZZLE)
  #define SWITCHING_NOZZLE_SERVO_NR 0
  //#define SWITCHING_NOZZLE_E1_SERVO_NR 1          // If two servos are used, the index of the second
  #define SWITCHING_NOZZLE_SERVO_ANGLES { 107, 58 }   // Angles for E0, E1 (single servo) or lowered/raised (dual servo)
#endif

/**
 * Z Servo Probe, such as an endstop switch on a rotating arm.
 */
#define Z_PROBE_SERVO_NR 1       // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES { 82, 107 } // Z Servo Deploy and Stow angles

#define DEACTIVATE_SERVOS_AFTER_MOVE
```
The head is a slight modification of this design.
https://www.thingiverse.com/thing:4441387

If there is some setting I have skipped over that would enable this please let me know.

thekingisback-code, 2021-11-18:
I’ve got a pretty similar setup, let me get home in a few days and I’ll share the code :) 
There’s also a big where you can’t set the second hotend offset correctly, have you encountered that or you haven’t made it that far?

kj4sst, 2021-11-18:
I was working ahead trying to get other parts working while the servo lock wasn't working and I found that if you were all the way at x = 0 and you switched tool heads, requiring the head to move more negative in the x direction, the head would not move. My temporary solution was to set the min x position to 0-(tool offset), in my case -40.

thekingisback-code, 2021-12-01:
Shoot, totally forgot about this, got an email I can contact you?

kj4sst, 2021-12-01:
You can contact me at joemcgarbage@gmail.com

kj4sst, 2021-12-05:
@thekingisback-code You can contact me at joemcgarbage@gmail.com

kj4sst, 2021-12-10:
@thekingisback-code Any chance I could get the code? Thanks in advance.

thinkyhead, 2022-04-21:
Using a single servo to switch between E0, E1, Probe will require some extra cleverness, so that (for example) switching away from the probe may return to E1 instead of E0… Though right now Marlin does a `T0` before probing, so you'd just need the probe to specify its own angle for "deployed" and the `E0` angle for "stowed", make the servo indices the same, and comment out any sanity-checks that prevent compile.