MarlinFirmware/Marlin #23126: Fysetc S6 V2 + Fysetc (FTDI EVE) 81050 display + Marlin 2.0.9.2 does not compile

maralb1970, 2021-11-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.0.9.2 does not compile with Fysetc FTDI EVE 81050 enabled.  Using a Fysetc S6 V2.0 controller. With just taking the basic confg + config_adv files; enabling the Fysetc S6, and Fysetc 81050 (+S6 pinmap) It throws the following error:

Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../language/language_en.h:148:24: error: redefinition of 'const char Language_en::MSG_MESH_VIEW []'

After Outcommenting this line in the ....Language_en.h Marlin 2.0.9.2 does compile.

Trying Bugfix 2.0.9.2 throws the same error as above. When Outcommenting the same Line another issue appears:

c:/users/user0/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/FYSETC_S6_8000/src/src/lcd/extui/ui_api.cpp.o: in function `ExtUI::onStatusChanged(__FlashStringHelper const*)':
D:\LEO_Firmware_Projects\GitCompileTest\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui/ui_api.cpp:1081: multiple definition of `ExtUI::onStatusChanged(__FlashStringHelper const*)'; .pio/build/FYSETC_S6_8000/src/src/lcd/extui/ftdi_eve_touch_ui/ftdi_eve_extui.cpp.o:D:\LEO_Firmware_Projects\GitCompileTest\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui\ftdi_eve_touch_ui/ftdi_eve_extui.cpp:74: first defined here

I played with the //#define TOUCH_UI_USE_UTF8 in config_adv.h but this didn't solve anything either:

Marlin\src\lcd\extui\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\unicode.cpp:77:15: error: redefinition of 'FTDI::utf8_char_t FTDI::get_utf8_char_and_inc(char*&)'

What is going on here? I might scr3w something myself and if I did, my apologies for that. I am not a coder but a hardware dinosaur....I do understand some Motorola assmebly and low-level C code.......

### Bug Timeline

Trying to get an FTDI EVE (Fysetc 81050) display working in Marlin 2.0.9.2

### Expected behavior

I expect it to compile.

### Actual behavior

It does not compile.

### Steps to Reproduce

Set up Platform I/O to Fysetc S6 8000 Environment
Set up Config.h for Fysetc S6, set serial port 1 to -1and enable FTDI EVE
Set up Config_Adv.h for Fysetc 81050 screen and S6 pinm
[cofig_files.zip](https://github.com/MarlinFirmware/Marlin/files/7532888/cofig_files.zip)
ap

### Version of Marlin Firmware

Marlin 2.0.9.2

### Electronics

Fysetc S6 V2.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

maralb1970, 2021-11-19:
The problem is not specifcally related to the Fysetc display. Enabling any of the FTDI EVE based displays results in the same error. There seems to be a conflict between 2 language header files:

\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language_en.h

and

\Marlin\src\lcd\language\language_en.h

ellensp, 2021-11-24:
Give this PR a try

maralb1970, 2021-11-24:
Thanks! Outcommenting these lines as done in this PR certainly seems to help. Does it affect any functionality? On the desk it seems to run fine, I will do some "real life" testing soon.
[image]

ellensp, 2021-12-14:
This PR has been merged. closing

rusviktorger, 2022-01-23:
English
Hi
I also have the same error.
But I don't understand what you mean with (Give this PR a try).
I have now downloaded the latest Marlin firmware (CONFIGURATION_H_VERSION 02000903).
And only changed the most necessary.: 

Default_envs = FYSETC_S6_8000
#define SERIAL_PORT 1
#define MOTHERBOARD BOARD_FYSETC_S6_V2_0
#define TOUCH_UI_FTDI_EVE
#define LCD_FYSETC_TFT81050 // FYSETC with 5" (800x480)
#define S6_TFT_PINMAP // FYSETC S6 pin mapping

But it doesn't work. 

[Neues Textdokument (2).txt](https://github.com/MarlinFirmware/Marlin/files/7920369/Neues.Textdokument.2.txt)

German
Hallo
ich habe auch den selben fehler.
Aber ich verstehe nicht was ihr mein mit (Give this PR a try).
Ich habe mir jetzt aktuellste Marlin firmware runtergeladen (CONFIGURATION_H_VERSION 02000903).
Und nur das nötigste verändert.:

Default_envs = FYSETC_S6_8000
#define SERIAL_PORT 1
#define MOTHERBOARD BOARD_FYSETC_S6_V2_0
#define TOUCH_UI_FTDI_EVE
#define LCD_FYSETC_TFT81050 // FYSETC with 5" (800x480)
#define S6_TFT_PINMAP // FYSETC S6 pin mapping

Aber es funktioniert nicht.

[Neues Textdokument (2).txt](https://github.com/MarlinFirmware/Marlin/files/7920369/Neues.Textdokument.2.txt)

ellensp, 2022-01-23:
https://github.com/MarlinFirmware/Marlin/pull/23424 added a global  `LSTR MSG_PLEASE_WAIT = _UxGT("Please wait...");` 
breaking this screen, again  
Just comment out `PROGMEM Language_Str MSG_PLEASE_WAIT              = u8"Please wait...";`
from Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language_en.h

rusviktorger, 2022-01-23:
English
Thank you for your prompt reply.
I now have in language_en.h (Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language_en.h)
The line( 119 //PROGMEM Language_Str MSG_PLEASE_WAIT = u8"Please wait...";) is commented out 

What do you mean by (added a global LSTR MSG_PLEASE_WAIT = _UxGT("Please wait...");)
Where should I paste it? Maybe him? (Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language_en.h)
How should I insert it, maybe like this? (#define LSTR MSG_PLEASE_WAIT = _UxGT("Please wait..."); 

German
danke für die schnelle antwort.
Ich habe jetzt im language_en.h (Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language_en.h)
Die zeille( 119 //PROGMEM Language_Str MSG_PLEASE_WAIT              = u8"Please wait...";) auskommentiert

Was meinen Sie mit (added a global LSTR MSG_PLEASE_WAIT = _UxGT("Please wait...");)
Wohin soll ich die einfügen? Vielleicht ihn? (Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language_en.h)
Wie soll ich die einfügen, vielleicht so?? (#define LSTR MSG_PLEASE_WAIT = _UxGT("Please wait...");

ellensp, 2022-01-23:
You don't insert anything, just delete the line from Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language_en.h.

rusviktorger, 2022-01-23:
English

A big thank you for the help
The display works.

Unfortunately, I found that not everything is supported, for example:
-AUTO_BED_LEVELING_BILINEAR (works limited)
-AUTO_BED_LEVELING_UBL (feature not at all)
-PSU_CONTROL (cannot be selected in the display)

I have now set AUTO_BED_LEVELING_BILINEAR.
Try to print something tomorrow.
Do you know if there is a firmware update for the display?
So the functions mentioned above work 

German

Ein großes Dankeschön für die Hilfe
Das Display funktioniert.

Leider habe ich festgestellt das nicht alles unterstützt wird zum Beispiel:
-AUTO_BED_LEVELING_BILINEAR (funktioniert eingeschränkt)
-AUTO_BED_LEVELING_UBL (Funktion überhaupt nicht)
-PSU_CONTROL (ist nicht im Display wählbar)

Ich habe jetzt AUTO_BED_LEVELING_BILINEAR eingestellt.
Versuche morgen was zu drucken.
Wissen Sie vielleicht ob es eine Firmware update für das Display gibt?
Sodas die oben genahten Funktionen funktionieren

ellensp, 2022-01-24:
All touch screen are limited. 
They support just enough feature that the original printer had and that is it.

closed 2022-01-24 by ellensp

maralb1970, 2022-01-24:
Just wondering as I am not a software engineer: is outcommenting these lines in the FTDI_language_en.h file a "fix" or a "bush fix"? Is it affecting functionality? Are the UTF8 string definitions in the FTDI_language_en.h file an addition to those defined in the other Marlin_language_EN.h file? Or are these 2 seperate worlds?

ellensp, 2022-01-24:
after the fix. The definition just moved from one file to another.

maralb1970, 2022-01-25:
Clear! Thank you ellensp :-)