MarlinFirmware/Marlin #23138: Compiling fails when Enabling HiPrecy DGUS LCD in Marlin 2.0.9.2

maralb1970, 2021-11-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.0.9.2 fails to compile after enabling HiPrecy DGUS LCD. Using the basic configuration files with just a Fysetc S6 V2.0 controller selected and the Hiprecy DGUS LCD enabled.  The following errors are the result:

In file included from Marlin\src\lcd\extui\dgus\hiprecy\../../../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\extui\dgus\hiprecy\DGUSDisplayDef.cpp:25:
Marlin\src\lcd\extui\dgus\hiprecy\DGUSDisplayDef.cpp:404:16: error: 'VP_FAN2_CONTROL' was not declared in this scope; did you mean 'VP_FAN1_CONTROL'?
Marlin\src\lcd\extui\dgus\hiprecy\DGUSDisplayDef.cpp:405:16: error: 'VP_FAN2_STATUS' was not declared in this scope; did you mean 'VP_FAN1_STATUS'?
Marlin\src\lcd\extui\dgus\hiprecy\DGUSScreenHandler.cpp:88:20: error: 'MKSLCD_SCREEN_PAUSE' was not declared in this scope; did you mean 'DGUSLCD_SCREEN_UNUSED'?

### Bug Timeline

Trying to run Marlin 2.0.9.2 from 2.0.5.1

### Expected behavior

I expect it to compile

### Actual behavior

It does not compile

### Steps to Reproduce

Define Fysetc S6 V2.0 controller in config
Define serial port -1 and set 115200 baudrate
#define DGUS_LCD_UI_HIPRECY
Set Fysetc S6 08000 environment in PlatformIO.ini
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7542273/configs.zip)

### Version of Marlin Firmware

2.0.9.2

### Printer model

HiPrecy Leo

### Electronics

Fysetc S6 V2.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

ellensp, 2021-11-16:
HiPrecy display is hard coded for 2 fans 
This controller has 3 fans and breaks it...

ellensp, 2021-11-16:
So the question becomes does it really not support more fan or not..
In Marlin/src/lcd/extui/dgus/hiprecy/DGUSDisplayDef.h  
enable
//constexpr uint16_t VP_FAN2_CONTROL = 0x2204;
and
//constexpr uint16_t VP_FAN2_STATUS = 0x3304;

This  compiles
But you need to check if the displays 3 fan controls and they work or not.  

If they do, we can update the code so it adds the fans as needed
If it doesn't we can make the code have an upper limit of 2 fans and not break.

maralb1970, 2021-11-16:
Thanks, that does ring a bell. The Hiprecy DGUS is built for the Hiprecy Leo printer that came originally with a Fysetc F6 V1.4 8 bit controller which also had 3 fan controllers. The FAN2 errors are my mistake: I forgot to enable the Controllerfan in config_adv.h

After enabling the controllerfan and assigning a pin for the controllerfan the Fan2 errors vanished. No issues there and no need to take action here.

Now only the "MKS error" remains:

Marlin\src\lcd\extui\dgus\hiprecy\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_ResumePauseAbort(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\dgus\hiprecy\DGUSScreenHandler.cpp:88:20: error: 'MKSLCD_SCREEN_PAUSE' was not declared in this scope; did you mean 'DGUSLCD_SCREEN_UNUSED'?
   88 |         GotoScreen(MKSLCD_SCREEN_PAUSE);
      |                    ^~~~~~~~~~~~~~~~~~~
      |                    DGUSLCD_SCREEN_UNUSED
*** [.pio\build\FYSETC_S6_8000\src\src\lcd\extui\dgus\hiprecy\DGUSScreenHandler.cpp.o] Error 1

maralb1970, 2021-11-16:
Updated config_adv.h....

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7544689/configs.zip)

ellensp, 2021-11-16:
This errot only happens when you enable SDSUPPORT

Give this a try in Marlin/src/lcd/extui/dgus/hiprecy/DGUSScreenHandler.cpp
change         GotoScreen(MKSLCD_SCREEN_PAUSE);
to        GotoScreen(DGUSLCD_SCREEN_SDPRINTMANIPULATION);

maralb1970, 2021-11-16:
After applying these changes Marlin 2.0.9.2 succesfully compiles. However, the display is not showing the "homescreen" anymore. It only executes the bootscreen animation and after this it freezes at the last image of the boot animation.

I went back into history and tried Marlin 2.0.8. This results and a bunch of different errors and fails to compile.

I tried Marlin 2.0.7.2. This compiled successfully and it shows the "homescreen" correctly after executing the bootscreen animation.

thisiskeithb, 2021-11-16:
> After applying these changes Marlin 2.0.9.2 succesfully compiles.

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) & apply the patch to test with the latest code.

maralb1970, 2021-11-16:
It shows the same behaviour. With both Marlin 2.0.9.2 and BugFix 2.0.9.2 the Homescreen not appears after the bootscreen animation and both stalling showing the last image of the bootscreen animation.

maralb1970, 2021-11-17:
I tested both DGUS_LCD_UI_FYSETC and DGUS_LCD_UI_RELOADED. Both options build/compile successfully in BugfixMarlin2.0.9.2 and they are both running without issues. Both are showing a bootscreen followed by the Homescreen after powering up. (Of course, I had to upload the specific datasets to the DGUS display in both cases). Only the Hiprecy DGUS UI is affected in Marlin 2.0.9.2

NoviceArtisan, 2022-12-04:
Not a god damn duplicate

closed 2026-10-20 by thinkyhead