MarlinFirmware/Marlin #23139: I can not compile Marlin 2 on VXCode PlatformIO

manuel1414, 2021-11-16:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

[BUG] During compilation I get this error and I not know how resolve:

`> Executing task: platformio run --target clean <

Error: Traceback (most recent call last):
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 129, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1254, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1297, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 25, in <module>
    from platformio.commands.device.command import device_monitor as cmd_device_monitor
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/platformio/commands/device/__init__.py", line 15, in <module>
    from platformio.commands.device.filters.base import DeviceMonitorFilter
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/platformio/commands/device/filters/base.py", line 15, in <module>
    from serial.tools import miniterm
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/serial/tools/miniterm.py", line 16, in <module>
    from serial.tools.list_ports import comports
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/manuelcastellanos/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "platformio 'run', '--target', 'clean'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

### Bug Timeline

new

### Expected behavior

Compile Marlin 2.0.7.2

### Actual behavior

Error on python scripts.

### Steps to Reproduce

Compile Marlin

### Version of Marlin Firmware

2.0.7.2

### Printer model

Trimplot A3

### Electronics

Mega 2560/Ramps 1.4

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

Personal Slicer and host, for macOS.

Tannoo, 2021-11-16:
> Did you test the latest bugfix-2.0.x code?
No, but I will test it now!

And the result?

Tannoo, 2021-11-16:
The latest stable version is 2.0.9.2.
I am sure my old version compiled just fine.

I don't believe this is a Marlin issue.
This is either a bad Python or PlatformIO install.

ellensp, 2021-11-16:
Python pyserial issue https://github.com/pyserial/pyserial/issues/509#issuecomment-726966076

closed 2021-11-16 by ellensp