MarlinFirmware/Marlin #23146: [FR] MAX31865 support for heated bed

Justase, 2021-11-16:
### Is your feature request related to a problem? Please describe.

Hello,
My printer heated bed is made out of 8mm casted aluminium that has PT100 sensor inserted to it, and a heated silicone pad with standard 100k thermistor. My intention was to wire PT100 to MAX31865 adapter board and use it as a main bed sensor. While Marlin supports MAX31865 for the hotend, it cannot be linked to the bed temp. 
In general, the MAX31865 support is already there, the only thing it needs to be linked to the bed. Help on how to tweak firmware to enable this link is appreciated.

### Are you looking for hardware support?

main board is BTT Octopus, 
the MAX31865 board is connected to SPI3

### Describe the feature you want

to support MAX31865 in Marlin not only for hotend, but for bed temp too, or as any temperature input.

ansonl, 2023-03-21:
@Justase Did you ever find a work around for this? I am trying to use MAX31865 for my bed temp as well.

Justase, 2023-03-21:
No, I haven’t looked further into this since early 2022.

dbuezas, 2023-11-06:
I managed to make this work.

I'm upgrading an Ultimaker 2 (which has the pt100 built in the heated bed) to with an SKR3 main board.

I intend to make a PR, but the max31865 code makes such aggressive use of preprocessor macros and mixes in support for other max models, that it is near impossible to shove this in without making a new dedicated function for the bed thermistor. That explains why support isn't there yet.

I'll open the PR once I finish the upgrade and confirm everything works.

thisiskeithb, 2023-11-19:
Added in https://github.com/MarlinFirmware/Marlin/pull/26441

closed 2023-11-19 by thisiskeithb