MarlinFirmware/Marlin #23147: [BUG] (bug summary) #define LASER_FEATURE make error compile with temp_bed

filipdns, 2021-11-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Only with uncomment #define LASER_FEATURE  and compile I received error

`Marlin\src\module\temperature.cpp: In static member function 'static void Temperature::updateTemperaturesFromRawValues()':
Marlin\src\module\temperature.cpp:2107:21: error: 'temp_bed' was not declared in this scope
     if (TP_CMP(BED, temp_bed.raw, maxtemp_raw_BED)) max_temp_error(H_BED);
                     ^
Marlin\src\module\temperature.cpp:2105:45: note: in definition of macro 'TP_CMP'
   #define TP_CMP(S,A,B) (TEMPDIR(S) < 0 ? ((A)<(B)) : ((A)>(B)))
                                             ^
Marlin\src\module\temperature.cpp:2107:21: note: suggested alternative: 'temp_dir'
     if (TP_CMP(BED, temp_bed.raw, maxtemp_raw_BED)) max_temp_error(H_BED);
                     ^
Marlin\src\module\temperature.cpp:2105:45: note: in definition of macro 'TP_CMP'
   #define TP_CMP(S,A,B) (TEMPDIR(S) < 0 ? ((A)<(B)) : ((A)>(B)))
                                             ^
Marlin\src\module\temperature.cpp:2107:35: error: 'maxtemp_raw_BED' was not declared in this scope
     if (TP_CMP(BED, temp_bed.raw, maxtemp_raw_BED)) max_temp_error(H_BED);
                                   ^
Marlin\src\module\temperature.cpp:2105:49: note: in definition of macro 'TP_CMP'
   #define TP_CMP(S,A,B) (TEMPDIR(S) < 0 ? ((A)<(B)) : ((A)>(B)))
                                                 ^
Marlin\src\module\temperature.cpp:2108:9: error: 'temp_bed' was not declared in this scope
     if (temp_bed.target > 0 && TP_CMP(BED, mintemp_raw_BED, temp_bed.raw)) min_temp_error(H_BED);
         ^~~~~~~~
Marlin\src\module\temperature.cpp:2108:9: note: suggested alternative: 'temp_dir'
     if (temp_bed.target > 0 && TP_CMP(BED, mintemp_raw_BED, temp_bed.raw)) min_temp_error(H_BED);
         ^~~~~~~~
         temp_dir
Marlin\src\module\temperature.cpp:2108:44: error: 'mintemp_raw_BED' was not declared in this scope
     if (temp_bed.target > 0 && TP_CMP(BED, mintemp_raw_BED, temp_bed.raw)) min_temp_error(H_BED);
                                            ^
Marlin\src\module\temperature.cpp:2105:45: note: in definition of macro 'TP_CMP'
   #define TP_CMP(S,A,B) (TEMPDIR(S) < 0 ? ((A)<(B)) : ((A)>(B)))
                                             ^
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\temperature.cpp:30:
Marlin\src\module\temperature.cpp: In static member function 'static void Temperature::update_raw_temperatures()':
Marlin\src\module\temperature.cpp:2895:31: error: 'temp_bed' was not declared in this scope
   TERN_(HAS_TEMP_ADC_BED,     temp_bed.update());
                               ^
Marlin\src\module\../inc/../core/macros.h:607:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\module\../inc/../core/macros.h:235:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:234:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:233:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
                             ^~~~~
Marlin\src\module\temperature.cpp:2895:3: note: in expansion of macro 'TERN_'
   TERN_(HAS_TEMP_ADC_BED,     temp_bed.update());
   ^~~~~
Marlin\src\module\temperature.cpp:2895:31: note: suggested alternative: 'temp_hotend'
   TERN_(HAS_TEMP_ADC_BED,     temp_bed.update());
                               ^
Marlin\src\module\../inc/../core/macros.h:607:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\module\../inc/../core/macros.h:235:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:234:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:233:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
                             ^~~~~
   TERN_(HAS_TEMP_ADC_BED,     temp_bed.update());
   ^~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\mega2560\src\src\module\temperature.cpp.o] Error 1`

### Version of Marlin Firmware

Marlin CONFIGURATION_H_VERSION 02000903

### Bed Leveling

ABL Linear grid

### Your Slicer

Other (explain below)

### Additional information & file uploads

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/7556505/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/7556506/Configuration.txt)

GMagician, 2021-11-17:
Do you really have a bed in your machine?

filipdns, 2021-11-17:
Yes, the error appear only with #define LASER_FEATUR, without that all is
working fine

thisiskeithb, 2021-11-17:
The config you attached is very out of date and does not build. Please attach the actual config you used with `bugfix-2.0.x`.:

`#error "TMC2208 or TMC2209 on Z2 requires Z2_HARDWARE_SERIAL or Z2_SERIAL_(RX|TX)_PIN."`

GMagician, 2021-11-18:
I suspect that laser will use bed heating output. If your machine has both (laser and heated bed) I suspect you have to define some new output

GMagician, 2021-11-18:
@thisiskeithb I confirm that no ramps configuration support laser and bed. Maybe a sanity check must be added somewhere because in such cnf HAS_HEATED_BED is not set (no output defined) but THERMAL_PROTECTION_BED is (because of bed sensor definition), hence the error!

filipdns, 2021-11-18:
> 
> 
> The config you attached is very out of date and does not build. Please attach the actual config you used with `bugfix-2.0.x`.:
> 
> `#error "TMC2208 or TMC2209 on Z2 requires Z2_HARDWARE_SERIAL or Z2_SERIAL_(RX|TX)_PIN."`

for solve this I add at the end in pins_MKS_GEN_L_V21.h

`#ifndef Z2_STEP_PIN
#define Z2_STEP_PIN E1_STEP_PIN
#define Z2_DIR_PIN E1_DIR_PIN
#define Z2_ENABLE_PIN E1_ENABLE_PIN
#endif`

filipdns, 2021-11-18:
> 
> 
> I suspect that laser will use bed heating output. If your machine has both (laser and heated bed) I suspect you have to define some new output

I'm searching for change this output because a have one not used but I can found where the laser output pin is set

GMagician, 2021-11-18:
Question is still: do you have an heated bed? Because solution may vary depending on this

Laser uses a servo but when LASER_FEATURE is enabled it will not define the bed output, hence the error.

Such features defines FET_ORDER_SF (spindle & fan) used by pins_RAMPS.h.

If you really have an heated bed you may add, at pins_MKS_GEN_L_V21.H:
#define HEATER_BED_PIN MOSFET_B.
Because A is used by heater0 and C by fan

if you don't have the heated bed, just remove sensor pin definition:
#define TEMP_SENSOR_BED 0

GMagician, 2021-11-18:
however you need #23150 to compile

closed 2021-11-27 by filipdns