MarlinFirmware/Marlin #23154: [BUG] (mks_robin_nano35)

PhyroXia, 2021-11-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

From today i have this issue while compiling, i changed nothing and 6 days before it worked fine. 

When i undefine FILAMENT_RUNOUT_SENSOR i can compile but i need that FILAMENT_RUNOUT_SENSOR

In file included from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/tft_io/tft_io.h:29,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/tft_io/touch_calibration.h:25,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/marlinui.h:38,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/trinamic.h:35,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/indirection.h:44,
                 from Marlin\src\gcode\feature\runout\../../../feature/../module/stepper.h:47,
                 from Marlin\src\gcode\feature\runout\../../../feature/runout.h:31,
                 from Marlin\src\gcode\feature\runout\M412.cpp:28:
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/tft_io/../../HAL/STM32/tft/tft_fsmc.h:24:10: fatal error: ../../../inc/MarlinConfig.h: No such file or directory
   24 | #include "../../../inc/MarlinConfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\mks_robin_nano35\src\src\gcode\geometry\M206_M428.cpp.o
*** [.pio\build\mks_robin_nano35\src\src\gcode\feature\runout\M412.cpp.o] Error 1

### Bug Timeline

18.11.2021

### Expected behavior

Should compile without error.

### Actual behavior

Stops compiling when FILAMENT_RUNOUT_SENSOR is defined

### Steps to Reproduce

Start Compiling when FILAMENT_RUNOUT_SENSOR is defined

### Version of Marlin Firmware

STRING_DISTRIBUTION_DATE "2021-11-12" "bugfix-2.0.x"

### Printer model

Fokoos Odin 5 F3

### Electronics

Stock

### Add-ons

none

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

tpruvot, 2021-11-18:
i see only two ".../..." in current marlin code...
https://github.com/descipher/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/tft_io/tft_io.h#L24

ellensp, 2021-11-18:
Cannot reproduce
Probably because steps to reproduce is woefully lacking in details.

Tested on current marlin bugfix 

platformio.ini
default_envs = mks_robin_nano35

Configuration.h
#define SERIAL_PORT 1
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define FILAMENT_RUNOUT_SENSOR
#define NOZZLE_PARK_FEATURE

Configuration_adv.h
#define EMERGENCY_PARSER
#define ADVANCED_PAUSE_FEATURE

PhyroXia, 2021-11-19:
> Cannot reproduce Probably because steps to reproduce is woefully lacking in details.
>
> Tested on current marlin bugfix
>
> platformio.ini default_envs = mks_robin_nano35
>
> Configuration.h #define [...]

I Tested also with the current Marlin Bug fix and it works again but i Wonder why the Marlin Bugfix from 12.11.21 stopped working without any Changes

closed 2021-11-20 by ellensp