MarlinFirmware/Marlin #23155: [BUG] say busy after pause (dgus reload from desuu firmware ) with skr14.T

labotecno, 2021-11-18:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

i have problem with pause with dgus screen from cr10 s pro , and i have skr14T , seem to have the same bug like this [https://github.com/Desuuuu/Marlin/issues/25](url)
when i try to pause i cant resume or printer dont make nothing . 
[https://github.com/Desuuuu/DGUS-reloaded](url)
and [https://github.com/Desuuuu/Marlin](url)

### Expected behavior

resume after pause 

### Version of Marlin Firmware

2.0.9.1

### Printer model

cr10spro

### Electronics

skr1.4T

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

ellensp, 2021-11-18:
We cannot support 3rd party forks of Marlin
This instant someone makes a fork of marlin they are responsible for the support of that fork.

labotecno, 2021-11-18:
ok bt marlin have #define DGUS_LCD_UI_RELOADED ? or not ? so making sense is part of marlin

ellensp, 2021-11-18:
If you use real marlin source code from this github repository and issue exists, then can re open.

labotecno, 2021-11-19:
> If you use real marlin source code from this github repository and issue exists, then can re open.

yes i use bug fix with DGUS_LCD_UI_RELOADED and same issue .

closed 2022-01-29 by github-actions[bot]

Romain80200, 2022-02-01:
Hello, I've got the same problem with my longer lk4 pro.