MarlinFirmware/Marlin #23157: [BUG] Robin nano v1.3 compilation error timers_in_use

Disttrack, 2021-11-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

665 / 5000
Resultados de traducción
When I activate UART mode in TMC2208 drivers on Robin Nano V1.3 board (STM32F407VET6), I set the RX and TX pins I get an error like this:
Marlin\src\HAL\STM32\timers.cpp:322:40: error: static assertion failed: One or more timer conflict detected. Examine "timers_in_use" to help identify conflict.
  322 | static_assert(verify_no_timer_conflicts(), "One or more timer conflict detected. Examine \"timers_in_use\" to help identify conflict.");
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~^~
*** [.pio\build\mks_robin_nano_v1_3_f4\src\src\HAL\STM32\timers.cpp.o] Error 1

I've tried some of the solutions for other boards but they won't work. 

### Expected behavior

Compile sucesfully

### Steps to Reproduce

Marlin\src\HAL\STM32\timers.cpp:322:40: error: static assertion failed: One or more timer conflict detected. Examine "timers_in_use" to help identify conflict.
  322 | static_assert(verify_no_timer_conflicts(), "One or more timer conflict detected. Examine \"timers_in_use\" to help identify conflict.");
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~^~
*** [.pio\build\mks_robin_nano_v1_3_f4\src\src\HAL\STM32\timers.cpp.o] Error 1

### Version of Marlin Firmware

2.0.9.2

### Printer model

Kingroon KP3S

### Electronics

Kingroon v1.2 STM32F407VET6

### Additional information & file uploads

[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/7569769/Marlin-config.zip)

cscomputers, 2021-11-20:
Hello! It solve for me Commenting the line or add an condition below for solve this on Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO_common.h:

// #define TEMP_TIMER                             5

OR add an condition....

[image]

And before this, works great!

[image]

Disttrack, 2021-11-23:
> Hello! It solve for me Commenting the line or add an condition below for solve this on Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO_common.h:
> 
> // #define TEMP_TIMER 5
> 
> OR add an condition....
> 
> [image]
> 
> And before this, works great!
> 
> [image]

Thank you! I have compiled commenting on the line you have indicated!

closed 2021-11-23 by Disttrack

cscomputers, 2021-11-23:
Oh great! I’m happy to help.