MarlinFirmware/Marlin #23159: [BUG] BOTH macro doesn't work as expected

rondlh, 2021-11-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In Marlin\src\gcode\queue.cpp the BOTH function is used:
#if BOTH(RX_BUFFER_MONITOR, RX_BUFFER_SIZE)
The code following this statement is not included if RX_BUFFER_SIZE is anything else than 1 (e.g. 128).
This is easy to show by adding incorrect code behind the statement (e.g. "abc") and compiling the code while RX_BUFFER_MONITOR enabled (Configuration_adv.h).
It compiles without errors, which means the added incorrect code was not included/compiled.

### Bug Timeline

Don't know, present in 2.0.9.2 bugfix (2021-11-20)

### Expected behavior

RX buffer monitor code should be compiled/executed, but it's not included during compile time.

### Actual behavior

RX buffer monitor code is never executed because it was not included during compile time.

### Steps to Reproduce

1. Enable RX_BUFFER_MONITOR (Configuration_adv.h)
2. Set RX_BUFFER_SIZE 256 (Configuration_adv.h)
3. Add an incorrect statement in Marlin\src\gcode\queue.cpp behind the line #if BOTH(RX_BUFFER_MONITOR, RX_BUFFER_SIZE)
4. Check if the code compiles correctly, which should not be the case, this means that the code was not included and the RX buffer monitoring will not work.

### Version of Marlin Firmware

2.0.9.2 official release, tested too with the current bugfix

### Printer model

Custom, LPC1769 motherboard with BTT TFT35 V3.0

### Electronics

LPC1769 motherboard with BTT TFT35 V3.0

### Add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

This bug might affect lots of other areas in the Marlin code base.
The use of the BOTH function should be investigated.

ellensp, 2021-11-20:
`BOTH(OPT1, OPT2) | True if both of the listed options is enabled.`

Enabled is either true or false  0 or 1,  but  >1 is not true in macros.

This test is wrong. not the macro

This is what the Macro is doing, each line is equivalent of the line above

  //#if BOTH(RX_BUFFER_MONITOR, RX_BUFFER_SIZE)
  //#if ALL(,2)
  //#if ENABLED(,2)
  //#if DO(ENA,&&,,2)
  //#if (_DO_N(ENA,&&,NUM_ARGS(,2),,2))
  //#if __DO_N(ENA,&&,2,,2)
  //#if _DO_2(ENA,&&,,2)
  //#if (_ENA_1() && _ENA_1(2))
  //#if ( _ISENA(CAT(_IS,CAT(ENA_, ))) && _ISENA(CAT(_IS,CAT(ENA_, 2))))
  //#if ( _ISENA(CAT(_IS,ENA_)) && _ISENA(CAT(_IS,ENA_2)))
  //#if ( _ISENA(_ISENA_) && _ISENA(_ISENA_2))        <=   _ISENA_2 does not exist
  //#if ( _ISENA(~,1) && _ISENA())
  //#if ( IS_PROBE(~,1) && IS_PROBE())
  //#if ( SECOND(~,1, 0) && SECOND(, 0) )
  //#if ( 1 && 0)

ellensp, 2021-11-20:
The test should probably be the following
```CPP
  #if RX_BUFFER_SIZE > 0 && ENABLED(RX_BUFFER_MONITOR)
```

thisiskeithb, 2021-11-23:
#23160 has been merged.

closed 2021-11-23 by thisiskeithb