MarlinFirmware/Marlin #23162: [BUG] The fan does not turn off

Malderin, 2021-11-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Board: MKS Robin nano v.2
FW: Marlin 2.x bugfix

The model's cooling fan does not turn off when the printer is turned on (after the loading is complete).
The normal behavior was this:
1. The fan turns on when the board boots
2. After the download is complete, the fan turns off.
This came after:After https://github.com/MarlinFirmware/Marlin/pull/231
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7574647/Configuration.zip)
49

### Version of Marlin Firmware

Marlin 2.x bugfix

Portzal, 2021-11-20:
2 out of my 3 PMW fans stopped working with the recent Bugfix, part Cooling and driver cooling. Changed to 2.0.9.2 and all worked as expected

descipher, 2021-11-21:
@Malderin Have you tried building with mks_robin_nano35, I have not looked at the maple build env yet to see if it is Ok.

Portzal, 2021-11-21:
My board is a BTT Octopus Pro.

ellensp, 2021-11-21:
@Portzal please dont take over other peoples issues.
this is for a  MOTHERBOARD BOARD_MKS_ROBIN_NANO_V2 and is not the same issues as fans not working at all

ellensp, 2021-11-21:
@Portzal fans not working is already known an being worked on see https://github.com/MarlinFirmware/Marlin/issues/23094

Malderin, 2021-11-21:
> 

I have been using mks_robin_nano35_maple for a long time.
Now I tried to build the firmware with default_envs = mks_robin_nano35 and got a compilation error.
Perhaps the error is not related to this PR, I do not know.

```
Compiling .pio\build\mks_robin_nano35\src\src\HAL\STM32\watchdog.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\timers.cpp:26:
Marlin\src\HAL\STM32\timers.cpp: In function 'void SetTimerInterruptPriorities()':
Marlin\src\HAL\STM32\timers.cpp:192:44: error: 'setInterruptPriority' is not a member of 'SoftwareSerial'
  192 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |                                            ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:607:26: note: in definition of macro 'THIRD'
  607 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\HAL\STM32\../../inc/../core/macros.h:235:29: note: in expansion of macro '___TERN'
  235 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:234:29: note: in expansion of macro '__TERN'
  234 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:233:29: note: in expansion of macro '_TERN'
  233 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin\src\HAL\STM32\timers.cpp:192:3: note: in expansion of macro 'TERN_'
  192 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |   ^~~~~
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32\timers.cpp.o] Error 1
========================= [FAILED] Took 119.05 seconds =========================
 
Environment       Status    Duration
----------------  --------  ------------
mks_robin_nano35  FAILED    00:01:59.051
==================== 1 failed, 0 succeeded in 00:01:59.051 ====================
```

Malderin, 2021-11-21:
Clarification. The fan turns on and runs during boot and after switching on. However, from the menu you can then turn it on and off.

Portzal, 2021-11-21:
> @Portzal fans not working is already known an being worked on see https://github.com/MarlinFirmware/Marlin/issues/23094

Missed that one and thought this was a similar issue to mine. My bad.

descipher, 2021-11-21:
> I have been using mks_robin_nano35_maple for a long time. Now I tried to build the firmware with default_envs = mks_robin_nano35 and got a compilation error. Perhaps the error is not related to this PR, I do not know.

I would suggest a Clean All and close any previous tasks.
I can confirm there are issues with set_pwm_duty on the maple based __STM32F1__ HAL so it needs work.
#23125 will address HAL_STM32 builds which are in the mks_robin_nano35 environment.

descipher, 2021-11-21:
I need to cherry-pick the referenced PR to do a verification whether it does or does not have an issue in either env. I will update later.

Malderin, 2021-11-21:
> I would suggest a Clean All and close any previous tasks.

Yes you are right. Cleaning up temporary files helped resolve compilation error.
However, the model's cooling fan is no longer working at all.

thisiskeithb, 2021-11-21:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23094

closed 2021-11-21 by thisiskeithb

descipher, 2021-11-21:
> > I would suggest a Clean All and close any previous tasks.
> 
> Yes you are right. Cleaning up temporary files helped resolve compilation error. However, the cooling fan of the default_envs = mks_robin_nano35 model now does not work at all.

I that with PR #23125?

Malderin, 2021-11-21:
> Is that with PR #23125?

With this PR and default_envs = mks_robin_nano35 everything works well.

descipher, 2021-11-21:
> > Is that with PR #23125?
> 
> With this PR and default_envs = mks_robin_nano35 everything works well.

Thanks for confirming.

robbycandra, 2021-11-24:
> Clarification. The fan turns on and runs during boot and after switching on. However, from the menu you can then turn it on and off.

Yes…  I have made PR to fix this.