MarlinFirmware/Marlin #23164: [BUG] Prepare Menu is missing

muellerti, 2021-11-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

I am using the latest Marlin 2.0.9.2 Release and cannot find the prepare menu to change the filament. 
The offical marlin documentation says "The Prepare menu is only available when the machine is not printing." but it is still not there in idle, in my case. 
I found some older threads and one suggested to enable NOZZLE_PARK_FEATURE and 
ADVANCED_PAUSE_FEATURE but unfortunately it doesn't work.  
I am using the upgraded 4.2.7 Silent Board, everything else is stock. I also tried the bugfix. Please find my Configuration.h and Configuration_adv.h attached as zip.

Am I missing something?

Regards,
muellerti

### Version of Marlin Firmware

2.0.9.2

### Printer model

Ender 3 Pro

### Electronics

4.2.7 Silent Board

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7575022/Configuration.zip)

muellerti, 2021-11-20:
Update: When running the M600 manually in octoprint, it says command not found. Seems like it is missing the feature at all?

ellensp, 2021-11-21:
ADVANCED_PAUSE_FEATURE is the only requirement  to activate M600 gcode

muellerti, 2021-11-21:
Hello,
thanks for pointing that out. I switched back from bugfix to release and flashed it again. Now there is the "change filament" option but still no "prepare" menu. I have no clue why. For me it is acceptable because I only need the change filament feature. 

My configuration has both flags enabled:
NOZZLE_PARK_FEATURE and ADVANCED_PAUSE_FEATURE

closed 2021-11-21 by muellerti