MarlinFirmware/Marlin #23165: [FR] 2 PIN adresses for CONTROLLER_FAN_PIN

HendrikJan-5D, 2021-11-21:
### Describe the feature you want

To support rebasing 9axis branch onto current upstream MarlinFirmware/Marlin/bugfix-2.0.x
Do I test hardware with a BTT GTR V1.0 plus M5 V1.0 board with 11 TMC2226 drivers
With my multi Axis 5D printer.

To cool the stepper drivers on the motherboard I connected a fan to the FAN2_PIN.
The extension board BTT M5 has also 5 stepper drivers and connectors for extra fans.
Is it possible to control 2 PIN addresses with #define CONTROLLER_FAN_PIN.
e.g. FAN2_PIN and FAN3_PIN

Because I can't cool both boards with 1 fan

see  Configuration.adv.h

#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
#define CONTROLLER_FAN_PIN FAN2_PIN       // Set a custom pin for the controller fan

### Additional context

[image]

[image]

robbycandra, 2021-11-22:
@HendrikJan-5D, why don't you wire both fans in parallel?

HendrikJan-5D, 2021-11-22:
@robbycandra , so far I have connected both fans in parallel.
But on the BTT M5 V1.0 expansion board there are 5 fan connections,
that cannot be used because the software does not support them.

When I started building my printer 2 years ago, there was no Marlin Multi-Axis support.
But thanks to the collaboration with DerAndere1, we did get them upstream together.
And I can control for my printer 4 extra axes with G codes in degrees of rotation.
I also want to try to do this with the development of new slicer software.
By offering my printer to test the software of slicer developers.

First 5D print on my printer.
[image]

robbycandra, 2021-11-22:
@HendrikJan-5D, Great Project!. I am waiting for your new slicer software.

DerAndere1, 2021-12-11:
You could try the following changes (untested): change 
https://github.com/MarlinFirmware/Marlin/blob/aebf70cdae0683ed067a9d5aa0f3c93883f0d55a/Marlin/src/feature/controllerfan.cpp#L42
to this:
  SET_OUTPUT(CONTROLLER_FAN_PIN);
  SET_OUTPUT(CONTROLLER_FAN2_PIN);

and change
https://github.com/MarlinFirmware/Marlin/blob/aebf70cdae0683ed067a9d5aa0f3c93883f0d55a/Marlin/src/feature/controllerfan.cpp#L75-L82

to this:
```
    #if ENABLED(FAN_SOFT_PWM)
      thermalManager.soft_pwm_controller_speed = speed;
    #else
      if (PWM_PIN(CONTROLLER_FAN_PIN))
        set_pwm_duty(pin_t(CONTROLLER_FAN_PIN), speed);
      else
        WRITE(CONTROLLER_FAN_PIN, speed > 0);
      if (PWM_PIN(CONTROLLER_FAN2_PIN))
        set_pwm_duty(pin_t(CONTROLLER_FAN2_PIN), speed);
      else
        WRITE(CONTROLLER_FAN_PIN2, speed > 0);
    #endif
```
----
Also, change
https://github.com/MarlinFirmware/Marlin/blob/aebf70cdae0683ed067a9d5aa0f3c93883f0d55a/Marlin/src/module/temperature.cpp#L2321-L2323

to this:
```
  #if ENABLED(USE_CONTROLLER_FAN)
    INIT_FAN_PIN(CONTROLLER_FAN_PIN);
    INIT_FAN_PIN(CONTROLLER_FAN2_PIN);
  #endif
```

and change
https://github.com/MarlinFirmware/Marlin/blob/aebf70cdae0683ed067a9d5aa0f3c93883f0d55a/Marlin/src/module/temperature.cpp#L3168

to this:
```
          if (soft_pwm_controller.count <= pwm_count_tmp) {
            WRITE(CONTROLLER_FAN_PIN, LOW);
            WRITE(CONTROLLER_FAN2_PIN, LOW);
          }
```
----
Also, change 
https://github.com/MarlinFirmware/Marlin/blob/aebf70cdae0683ed067a9d5aa0f3c93883f0d55a/Marlin/src/inc/Conditionals_post.h#L2565

to this:
```
                     && CONTROLLER_FAN_PIN != FAN##F##_PIN \
                     && CONTROLLER_FAN2_PIN != FAN##F##_PIN \
```
----
Also, change
https://github.com/MarlinFirmware/Marlin/blob/aebf70cdae0683ed067a9d5aa0f3c93883f0d55a/Marlin/Configuration_adv.h#L509-L511

to the following:
```
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN FAN2_PIN // Set a custom pin for the controller fan
  #define CONTROLLER_FAN2_PIN FAN3_PIN // Set a custom pin for the controller fan 2
```

HendrikJan-5D, 2021-12-11:
Which marlin version did you use .
My files differ from your files.
I am currently using Marlin-10axis_RP1

[Marlin-10axis_PR1_7axisconfig 5.zip](https://github.com/MarlinFirmware/Marlin/files/7697627/Marlin-10axis_PR1_7axisconfig.5.zip)

DerAndere1, 2021-12-11:
The changes were based on current MarlinFirmware/Marlin/bugfix-2.0.x
Dont bother with testing it right now, I try to publish a new branch tomorrow that will have
-this feature (CONTROLLER_FAN2_PIN)
-support for up to 9 axes
-Axes can be either linear or rotational
-units for rotational axes will be degrees (not mm or inches)
-feedrate for moves involving only rotational axes will be in degrees per minute like in NIST RS274NGC / LinuxCNC

It would be great to have you on board for testing

DerAndere1, 2021-12-13:
Here is the feature branch: https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_dev 
I updated your configs and was able to compile with them. Your updated configs are attached to this comment. There is now a setting `HAS_ROTATIONAL_AXIS4 true`, `HAS_ROTATIONAL_AXIS5 true` etc. for each axis in Configuration.h. Please start testing with small, slow moves. For moves that involve linear axes (XYZ) as well as rotational axes, feedrate is now interpreted as the feedrate along the XYZ path (as if no rotational axe were moved). The rotational axes will move in a synchronized fashion. This might lead to extremely rapid movement of the rotational axes for moves that involve small changes in X/Y/Z positions but big positional changes along the A/B/C/U axes. Example:
`G1 X1 Y1 Z1 A360 B360 C360 F20 ; Caution with higher feedrates`
[Marlin2ForPipetBot_HendrikJan-5D.zip](https://github.com/MarlinFirmware/Marlin/files/7700279/Marlin2ForPipetBot_HendrikJan-5D.zip)
Also it would be nice if you could enable `INCH_MODE_SUPPORT` and test inch mode (G20) and metric mode (G21). In both modes, rotational axes should now be in the unit degrees (°).

HendrikJan-5D, 2021-12-13:
In which branch should I copy the modified config files you sent to me.
If I use the branch below, the compilation goes completely wrong.
https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_dev

HendrikJan-5D, 2021-12-17:
Due to the large mass of the x and y axes I could not develop sufficient speed.
Brought the driver voltage from 12 volts to 24 volts via a booster power converter.
Now the stepper motors run better and I can achieve much higher speeds.

Compilation goes well only 2 PROBLEMS.

[image]

The CONTROLLER_FAN2_PIN function works fine, thanks for the fix.

Your modification of the rotation axes did not seem to have made any changes .
As before, the speed of all axes is determined by the slowest feedrate of an axis
Which is determined by the DEFAULT_MAX_FEEDRATE value.
And they all arrive at the end position at the same time. ??????

---------------------------------------------------------

The inch mode G20 gives no difference in reading, see log.

>>>  G1 X20 Y20 Z240 A45 B45 C45 U45 F200
SENDING:G1 X20 Y20 Z240 A45 B45 C45 U45 F200
>>> M114
SENDING:M114
X:20.00 Y:20.00 Z:240.00 A:45.00 B:45.00 C:45.00 U:45.00 E:0.00 Count X:3200 Y:3200 Z:384000 A:40000 B:40000 C:64000 U:4800
>>> G20
SENDING:G20
>>> M114
SENDING:M114
X:20.00 Y:20.00 Z:240.00 A:45.00 B:45.00 C:45.00 U:45.00 E:0.00 Count X:3200 Y:3200 Z:384000 A:40000 B:40000 C:64000 U:4800
>>> G21
SENDING:G21
>>> M114
SENDING:M114
X:20.00 Y:20.00 Z:240.00 A:45.00 B:45.00 C:45.00 U:45.00 E:0.00 Count X:3200 Y:3200 Z:384000 A:40000 B:40000 C:64000 U:4800

Or should I read the degrees differently for the rotation axes.   ?????

DerAndere1, 2021-12-17:
> As before, the speed of all axes is determined by the slowest feedrate of an axis Which is determined by the DEFAULT_MAX_FEEDRATE value.

This is good news. In my test machine the rotational axes can turn indefinitely and I was under the impression that DEFAULT_MAX_FEEDRATE was not respected for the rotational axes in moves like this: 
```
G21
G1 X1 A2000
```
I will test again with lower DEFAULT_MAX_FEEDRATE for the I (A) axis

To test inch mode, try this:
```
G1 X50 Y50 Z50 A0 B0 C0; safe position
G91; incremental movement mode
G20; inch mode
G1 A4 ; rotates 4*2.4 degrees (previous behaviour) or rotates 4 degrees (fixed behaviour)
G1 B4
G1 C4
M114
G21; back to sanity with metric mode
G1 A-2; rotates 1 degree back (old and new behaviour)
G1 B-2
G1 C-2
M114
G90; absolute movement mode
```

HendrikJan-5D, 2021-12-17:
FYI: my  max feedrate settings are #define DEFAULT_MAX_FEEDRATE   { 200, 200, 15, 200, 200, 200, 200, 25 }

Which is not quite clear to me.
What has changed in the control of the axes due to your last change.

-------------------------------------

log test inch mode

>>> G1 X50 Y50 Z50 A0 B0 C0
SENDING:G1 X50 Y50 Z50 A0 B0 C0
>>> G91
SENDING:G91
>>> G20
SENDING:G20
>>> G1 A4
SENDING:G1 A4
>>> G1 B4
SENDING:G1 B4
>>> G1 C4
SENDING:G1 C4
>>> M114
SENDING:M114
X:50.00 Y:50.00 Z:50.00 A:4.00 B:4.00 C:4.00 U:0.00 E:0.00 Count X:8000 Y:8000 Z:80000 A:3556 B:3556 C:5689 U:0
>>> G21
SENDING:G21
>>> G1 A-2
SENDING:G1 A-2
>>> G1 B-2
SENDING:G1 B-2
>>> G1 C-2
SENDING:G1 C-2
>>> M114
SENDING:M114
X:50.00 Y:50.00 Z:50.00 A:2.00 B:2.00 C:2.00 U:0.00 E:0.00 Count X:8000 Y:8000 Z:80000 A:1778 B:1778 C:2844 U:0
>>> G90
SENDING:G90

DerAndere1, 2021-12-17:
Thanks for confirming that everything works for all axes.

> What has changed in the control of the axes due to your last change.

One bug that is now fixed is this: 
previously, in inch mode, whenever you specified a position `n` for rotational axes `A` like `G1 An`, the axis `A` was rotated `n * 25.4` degrees. With the new version the A axis is correctly rotated `n` degrees. Example:
```
G91; incremental movement mode
G20; inch mode
G1 A4 ; rotates 4*25.4 degrees (previous behaviour) or rotates 4 degrees (fixed behaviour)
```
This only affected inch mode, not metric mode. 

Second bug that is now fixed:
Move commands involving linear axes are executed quicker now, under some circumstances:
Lets assume that we have 3 primary linear axes X, Y, Z.
And three secondary linear axes U,V,W
And three rotary axes A, B, C
Lets call dX, dY, dZ, dA... the moved distances along the X, Y, Z, A... axes. Let F be the specified feedrate. Lets assume F is low enough that moves are not limited by `DEFAULT_MAX_FEEDRATE`. Then the time (T) that execution of a move command takes is:
 
old version: 
T = sqrt(dX^2 + dY^2 + dZ^2 + dA^2 + dB^2 + dC^2+ dU^2+ dV^2+ dW^2)/F

new version:
if any of X, Y, Z move (whether other axes are moved or not)
  T = sqrt(dX^2 + dY^2 + dZ^2)/F
else, if none of XYZ move but any secondary linear axes move (wether rotational axes are moved or not):
  T = sqrt(dU^2 + dV^2 + dW^2)/F
else, if only rotary axes are moved:
  T = sqrt(dA^2 + dB^2 + dC^2)/F

Thus, moves like the following should be done much quicker with the new version:
```
G1 X0.1 A360.0 F10
```

HendrikJan-5D, 2021-12-18:
Thanks for your explanation.

Do you have any more new software ideas for a multi-axis printer systems?
When will you get the latest changes upstream in Marlin/bugfix-2.0.x
 
Frohe Weihnachten und ein glückliches und gesundes 2022

DerAndere1, 2021-12-18:
Apart from the extra fan, Its now all in the open pull request 23112. Maybe I will open a CONTROLLER_FAN2_PIN pull request later. Maybe never, because only few people will need it, which does not justify the bloat in code an config size. There are too many alternatives: hardware solutions (connection of two fans in parallel with appropriate circuit, stand-alone fans that are always on) or software solutions (connecting a fan between GND and any free pin and toggling that pin with M43 to switch the fan on, ...). CONTROLLER_FAN2_PIN will become part of Marlin2ForPipetBot in my fork, though. You can close this issue, I guess. 

Regarding multi-axis-support, I plan to do bugfixes, and try to find people that can fix some code sections marked with `TODO (DerAndere)` for me. When someone sends me an implementation of 5 axis kinematics, I will try to get it merged into Marlin....

thisiskeithb, 2023-01-09:
Implemented in https://github.com/MarlinFirmware/Marlin/pull/24995

closed 2023-01-09 by thisiskeithb