MarlinFirmware/Marlin #23166: [BUG] Compile error when commenting out SPINDLE_LASER_USE_PWM

tbnobody, 2021-11-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

My spindle doesn't use a PWM control. So I commented out SPINDLE_LASER_USE_PWM in Configuration_adv.h

But then I get e.g. the following error message:
`Marlin\src\feature\../module/../feature/spindle_laser_types.h:38:13: error: 'SPEED_POWER_MAX' was not declared in this scope`

Imho this issue was introduced in this commit: https://github.com/MarlinFirmware/Marlin/commit/de4eed33e49b889b9a29e417c991e4564bbe634c#diff-c7a3fa925098d81e51840820898b66e3d3e48f121a83e4d0b6c65e5ab4f2fba0R3400 because SPEED_POWER_MAX is no only defined if SPINDLE_LASER_USE_PWM is enabled.

### Version of Marlin Firmware

2.0.9.2

descipher, 2021-12-23:
Please retest, the merge should have resolved it

tbnobody, 2021-12-24:
Yes its working again! Thank you!

closed 2021-12-24 by tbnobody