MarlinFirmware/Marlin #23168: [FR] Fan status/tacho monitoring

bugfixin, 2021-11-21:
### Is your feature request related to a problem? Please describe.

I recently had a print fail due to heat creep when one of the wires to my hotend fan broke. That fan has a (currently disconnected) tachometer monitoring pin. If Marlin could monitor that pin and take some form of action or alerting, it would be very helpful in making sure that doesn't happen again.

### Describe the feature you want

I'd like marlin to be capable of monitoring a (3-pin) fan's tachometer output to ensure that the hotend fan is still spinning as expected. This could be extended to support for monitoring other system fans or even parts-cooling fans, but the hotend fan is the most critical in my opinion. 

As a thought, you could set an interrupt on a pin to store the last time that the tachometer pin state changed, and occasionally check the difference between now and when that last happened to ensure that it hasn't been too long.

### Additional context

From what I understand, the tacho output on most fans is an open-drain, 2 pulse-per-revolution output. That means a pull-up resistor is needed, and the interrupt would trigger about once every 20 milliseconds for a 1500RPM fan, or once every 6 milliseconds for a 5000RPM fan. That seems relatively reasonable to me for monitoring a critical fan, but I'm not sure the amount of free cycles that would tie up for an average 8bit micro running marlin.

thisiskeithb, 2021-11-21:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/11642

thisiskeithb, 2021-11-21:
This is being worked on: https://github.com/MarlinFirmware/Marlin/pull/23086

closed 2021-11-21 by thisiskeithb