MarlinFirmware/Marlin #23176: [FR] Support BTT SKR 2 stm32f429

frogamic, 2021-11-23:
### Are you looking for hardware support?

Bigtree Tech SKR2 with stm32f429 - 

### Describe the feature you want

Support SKR2 with stm32f429 CPU.
Big Tree Tech are now shipping the SKR 2 controller with a stm32f429 cpu, they do not seem to label this as a new revision.
See note under description: https://www.aliexpress.com/item/1005003002226879.html

ellensp, 2021-11-23:
"From October 7th, the SKR 2 uses "ARM Cortex-M4 STM32F429VGT6 180MHz"." 

Gee thank BTT

ellensp, 2021-11-23:
Looks like BTT have done most of the work in this https://github.com/bigtreetech/SKR-2/blob/master/Firmware/Marlin-bugfix-2.0.9.2.x/ini/stm32f4.ini#L248

ellensp, 2021-11-23:
@frogamic Can you test this PR? Its basicaly just extrated from the BTT code, it should work. 
Just choose the new build environment BIGTREE_SKR_2_F429

frogamic, 2021-12-01:
@ellensp I can confirm from patching in your changes that this works with my F429 board.

ellensp, 2021-12-01:
Since the PR is created to implement this, im going to close the FR as done (just waiting on administration stuff)

closed 2021-12-01 by ellensp