MarlinFirmware/Marlin #23178: I3 Mega, PROBE_MANUALLY with ABL

Jumbo125, 2021-11-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I use marlin for my Ender.
Now i want create a new version for my Anycubic i3 Mega. It works, but the movement/travel from XY is very slow, when i go to the next probe_point.
Homing  works great. 
Only the probe_manually not work.

Thank you for your answer

### Steps to Reproduce

1. Disalbe Auto_bed_
2. set up XY_PROBE_FEEDRATE (100*60) higher
3. set HOMING_FEEDRATE_MM_M higher 
4. disable AUTO_BED_LEVELING_BILINEAR

### Version of Marlin Firmware

bugfix-2.0.x, example config from 2.0.8.1

### Printer model

i3 Mega

### Electronics

stock electronics

### Add-ons

none

### Bed Leveling

ABL Bilinear mesh

thisiskeithb, 2021-11-23:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Jumbo125, 2021-11-23:
okay thank you. i thought it was a issue

Jumbo125, 2021-11-25:
Hello

I asked my question in some Forums(rerap) diacard and more...
Unfortunly Nobody Had an answer. 
I think ist a Bug/Issue. Is it right?

Jumbo125, 2021-11-26:
@thisiskeithb 
I wrote my Problem in some Forums(reprap, stackoverflow, discord, ans some Others printed Forums. Nobody knows an answer! 

So i think it's an Issue.