MarlinFirmware/Marlin #23189: [FR] ADD STM32G0B1RET6 env and SKR MINI E3V3 BOARD added to main Marlin 2.0.9.2

nick4sharm, 2021-11-24:
### Is your feature request related to a problem? Please describe.

BLT firmware using bugfix-2.0.x provided by SKR github doesn't allow for Layer and Controller Fan to work properly.
Their basic firmware.bin makes them work but any other firmware creates this issue.  

### Are you looking for hardware support?

SKR MINI E3 V3

### Describe the feature you want

BLT firmware using bugfix-2.0.x provided by SKR github doesn't allow for Layer and Controller Fan to work properly.
Their basic firmware.bin makes them work but any other firmware creates this issue.

thisiskeithb, 2021-11-24:
> Their basic firmware.bin makes them work but any other firmware creates this issue.

Looks like users have to manually copy some files to the framework to build properly: https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1#note

Hopefully @bigtreetech has a PR ready for this 🙂

thisiskeithb, 2021-12-10:
Support has been added in https://github.com/MarlinFirmware/Marlin/pull/23283

closed 2021-12-10 by thisiskeithb