MarlinFirmware/Marlin #23196: [BUG] REPORT_FAN_CHANGE Not Always Reporting

rongoodman303, 2021-11-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

[RonGoodman303_Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7599558/RonGoodman303_Marlin.zip)

Using BTT SKR2 Rev B motherboard, with 3 PMW fans. 
Tested on Marlin bugfix 2.0.9.1, but see no reports or fixes in 2.0.9.2 for this issue.

Applicable specific fan related options set in Configuration_adv.h (attached in zip):

#define REPORT_FAN_CHANGE   // Report the new fan speed when changed by M106 (and others)

#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN FAN2_PIN           // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SPEED_MIN         0 // (0-255) Minimum speed. (If set below this value the fan is turned off.)
  #define CONTROLLERFAN_SPEED_ACTIVE    255 // (0-255) Active speed, used when any motor is enabled
  #define CONTROLLERFAN_SPEED_IDLE        0 // (0-255) Idle speed, used when motors are disabled
  #define CONTROLLERFAN_IDLE_TIME        60 // (seconds) Extra time to keep the fan running after disabling motors

#define FAN_KICKSTART_TIME 100 // Force start fan if initial fan speed is low.

#define E0_AUTO_FAN_PIN FAN1_PIN // Assign to hotend for auto fan turn on/off
// For F1, SKRV2 board uses FAN1_PIN for this PMW fan

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50  // Set your own auto fan on temp
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed

=====================================

All of the above settings work, as evidenced by fan activity. Controller fan turns on when steppers are active, and E0 Fan turns on when hotend gets to 50c.

However, E0 fan and Controller fan speeds are not reported via serial port to BTT TFT35, which always shows F1 speed as 0, and F2 speed as 0. 

Pronterface Command issued: M123 S5
Result:  echo:Unknown command: "M123 S5"

Pronterface Command issued: M106 P2 S0
Expected Result: Controller fan (F2) is on, and should turn off.
Result:  Ponterface says: SENDING:M106 P2 S0, but controller fan remains on, and no speed change is reported to BTT TFT35.

Fan 0 (parts cooling) works as expected:
Pronterface Command issued: M106 P0 S128
Expected Result: Patr cooling fan (F0) should report 50% to TFT35.
Result:  Ponterface says: SENDING:M106 P0 S128, and the speed change is correctly reported to BTT TFT35, which correctly reports F0 at 50%.

### Bug Timeline

New

### Expected behavior

I expected fans F1 and F2 to be controlled by M106 commands and reported via serial port to BTT TFT35, just line F0 is. See comments in bug report

### Actual behavior

Fans F1 ans F2 were not controlled by M106, and fan speed changes were not reported to display, like F0 was. 

### Steps to Reproduce

See notes in bug report

### Version of Marlin Firmware

v 2.0.9.1 bugfix

### Printer model

Longer LK5 Pro

### Electronics

BTT SKR2 main board with BTT TFT35 v3.0 display

### Add-ons

NA

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

thisiskeithb, 2021-11-25:
> Tested on Marlin bugfix 2.0.9.1, but see no reports or fixes in 2.0.9.2 for this issue.

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, be sure to update the TFT to the latest version from a few days ago.

rongoodman303, 2021-11-29:
OK. Updates as requested:

BTT TFT35 firmware was updated to latest release (BIGTREE_TFT35_V3.0._E3.27.x.bin), with updated configuration.ini, icons, etc.
Marlin was updated to latest 2.0.9.3 bugfix. In the config files: 
#define CONFIGURATION_H_VERSION 02000903
Devart Code Compare was used to move my working 2.0.9.1 SKR2 and printer options over to the version 2.0.9.3 bugfix Configuration.h and Configuration_adv.h files. 
The 2.0.9.3 build went well and new firmware loaded OK to the SKR2 board. 

My 2.0.9.1 and 2.0.9.3 config versions are included in the zip file here, as are copies of the TFT35 files after being implemented.

Results:

Controller fan seems to be properly reported on TFT35 as CtA fan. However, when its speed is changed from Pronterface, the new fan speed is not reported, and CtA fan is always reported at 100%

Pronterface Log:
>>> M710
SENDING:M710
  M710 S255 I0 A1 D60 ; (100% 0%)
>>> M710 S128
SENDING:M710 S128
>>> M710
SENDING:M710
  M710 S128 I0 A1 D60 ; (50% 0%)      <=== But TFT35 still says 100%
>>> M710 S255
SENDING:M710 S255
>>> M710
SENDING:M710
  M710 S255 I0 A1 D60 ; (100% 0%)

For M106 results:

>>> M106
SENDING:M106
M106 P0 S255
>>> M106 S245
SENDING:M106 S245
M106 P0 S245           <== and TFT35 reports F0 speed at 96%, which is correct.
>>> M106 S255
SENDING:M106 S255
M106 P0 S255

But for other Fan Indexes, no speed is reported:

>>> M106 P1
SENDING:M106 P1          <== Should this not be the extruder (E0_AUTO_FAN) fan? 
>>> M106 P2
SENDING:M106 P2
>>> M106 P3
SENDING:M106 P3

So apparently, the extruder cooling fan is not assigned to a fan index? Or its speed cannot be determined and reported?

So now the issue boils down to this::
1) Exactly which fans do the respective fan indexes in M106 represent? Needs online documentation plus comments in Configuration_adv.h
2) For the fans defined in Configuration_adv.h in the Extruder Fans section, what are the various fan indexes, how are they assigned, and how do they get controlled with M106?
3) When an extruder fan is turned on/off, as mine is at 50c, why is this not being reported as specified by:   
#define REPORT_FAN_CHANGE // Report the new fan speed when changed by M106 (and others)
4) Why is there no apparent consistency between M710 and M106 functionality?

[RonGoodman303_Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7618964/RonGoodman303_Marlin.zip)

rongoodman303, 2021-11-30:
Additional Fan Testing and Results:
Pronterface loaded and connected to SKR2.

Attempted fan control from TFT35.

When the PMW part cooling fan (F0 on SKR2) is activated or controlled via the TFT35, that fan speed change activity is properly echoed back to Pronterface:
M106 P0 S255
M106 P0 S0

When attempting to control other fans from the TFT35:

The fan identified on the the TFT as CtI properly controls the Controller fan (F2 on SKR2 board) just as it should, including speed changes, but nothing is ever echoed back to Pronterface for the fan speed changes.

For the fan identified on the the TFT as CtA, which is the hotend/nozzle fan, the TFT fails to control that fan at all. The fan cannot be set to full and started, and cannot be stopped if/when started by Marlin as a result of the 50c hotend temp trigger as set in Configuration_adv.h. Marlin correctly starts and stops that fan based on current hotend temp, but never reports that back to Pronterface or to the TFT.

rongoodman303, 2021-12-16:
I did all the updates to later releases and retested and posted new test results. Is anyone going to change status back to active and look into this?

closed 2022-03-24 by github-actions[bot]