MarlinFirmware/Marlin #23198: [BUG] Tronxy X5SA (Pro) needs _maple build to work

tommywienert, 2021-11-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, i really hope this time it is the right place, as i already reported at ABM,which seems to be wrong.
I got a new printer, Tronxy X5SA Pro and found the chitu firmware to be terrible, so i followed the guideline (https://github.com/MarlinFirmware/Configurations/tree/2.0.9.2/config/examples/Tronxy/X5SA) to build my own update.cbd to get Marlin onto it.
The Build succeeded, i put the update.cbd onto the SD card and turned the Power on, and got a series of two short beeps (5x) and then an "exitus" style never ending beep. I looked through everything and tried to change settings until i found a German Forum where several people told to use the _maple build instead of the normal one (https://drucktipps3d.de/forum/topic/tutorial-tronxy-x5sa-von-chitu-auf-marlin/?part=22#postid-173633). Stupid as i am i did not try that so far, but when i did, it worked first try. i changed nothing in vs code (though i am using Codium) except to build the _maple instead of the normal build. Since i got some help on the Marlin Discord and gave feedback that it works now, i got told to create a bug ticket so that you guys know of this.

Greetings
Tommy

### Bug Timeline

2021-11-23

### Expected behavior

Marlin is flashed successfully and boots up

### Actual behavior

The printer beeps short twice 5 times first and then beeps continously.
The printer mainboard is doing nothing visible anymore and a STM reflash of the Chitu FW backup is needed (or a working update.cbd)

### Steps to Reproduce

1. Get VS Code, platformio and Auto Build Marlin working
2. Use Configuration from https://github.com/MarlinFirmware/Configurations/tree/2.0.9.2/config/examples/Tronxy/X5SA (see attached zip)
3. Use platform.ini (see attached zip)
4. Click Build on chitu_f103
5. use the update.cbd from Marlin/.pio/build/chitu_f103/
6. the printer will double beep 5 times and then continuously beep
7. back in Auto Build Marlin click Build on chitu_f103_maple
8. use the update.cbd from Marlin/.pio/build/chitu_f103_maple/
9. the printer beeps twice for 6 times and then Marlin FW boots up automatically

### Version of Marlin Firmware

2.0.9.2

### Printer model

Tronxy X5SA Pro

### Electronics

everything "out of box"

### Add-ons

N/A

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

I did not Select Bed Leveling,Slicer, Host Software and can not attach M111 Log output as the printer won't boot with the wrong cbd file.
[Tronxy.zip](https://github.com/MarlinFirmware/Marlin/files/7600614/Tronxy.zip)

tommywienert, 2021-11-25:
[image]
[image]
attaching 2 more screenshots of Auto Build Marlin

ellensp, 2021-11-28:
Took a quick loo at this
the Maple env builds at address 08008800  but the non maple defaults to 08000000

If you edit ini/stm32f1.ini find this section
```
[env:chitu_f103]
platform             = ${common_stm32.platform}
extends              = stm32_variant
board                = genericSTM32F103ZE
board_build.variant  = MARLIN_F103Zx
build_flags          = ${stm32_variant.build_flags}
                       -DSTM32F1xx -DSTM32_XL_DENSITY
build_unflags        = ${stm32_variant.build_unflags}
extra_scripts        = ${stm32_variant.extra_scripts}
                       buildroot/share/PlatformIO/scripts/chitu_crypt.py
```

And update it to
```
[env:chitu_f103]
platform             = ${common_stm32.platform}
extends              = stm32_variant
board                = genericSTM32F103ZE
board_build.variant  = MARLIN_F103Zx
board_build.offset   = 0x8800
build_flags          = ${stm32_variant.build_flags}
                       -DSTM32F1xx -DSTM32_XL_DENSITY
build_unflags        = ${stm32_variant.build_unflags}
extra_scripts        = ${stm32_variant.extra_scripts}
                       buildroot/share/PlatformIO/scripts/chitu_crypt.py
```

This updates the start address, but more may be required..  

Can you give this a try?

tpruvot, 2021-11-28:
by the way, i dont think the F103ZE is XL density.... XL is for F & G chips cf https://www.st.com/resource/en/datasheet/stm32f103rg.pdf

tpruvot, 2021-11-28:
seems only used in maple fsmc defines, so it seems useless for the STM32 framework... you could replace it by STM32_HIGH_DENSITY but should be made automatically by the framework

tommywienert, 2021-11-28:
@ellensp That did the trick, thank you for taking a look
[image]

So, as this is my first reasonable bugticket in Github, how to proceed? shall i upload that change on a branch and create a pull request?

tpruvot, 2021-11-28:
i would say yes, but right now (today) all PRs will fail on the tests :p but should be ok tomorrow

tommywienert, 2021-11-28:
i won't be able to do that before tomorrow anyways ;) 
Should i also add the "XL vs High Density" thing as well ?

ellensp, 2021-11-28:
also try removing the -DSTM32_XL_DENSITY  as suggested by tpruvot. It shouldn't be needed

tommywienert, 2021-11-28:
@ellensp rebuilt and flashed without the parameter, worked as well

ellensp, 2021-11-28:
Two changes to make in the PR then. Time permitting.

ellensp, 2021-12-05:
This has been merged. Closing

closed 2021-12-05 by ellensp