MarlinFirmware/Marlin #23206: [FR] Multiple INA226 support

rflulling, 2021-11-26:
### Is your feature request related to a problem? Please describe.

Feature Request: Inspired by the addition of the I2C amperage checking function via a singular INA226. I would like to ask for the ability to monitor one chip or channel per Axis. INA226 have one per chip. INA3221 have 3 per chip. 

### Are you looking for hardware support?

INA226
INA3221

### Describe the feature you want

I would like to be able to use the INA226 or INA3221 to monitor the voltages and amperage of multiple axis. 

In particular this may be useful in monitoring CNC axis load. Normally to do so a shunt or clamp must be placed inline. An extra monitor installed at the CNC router motor 120V or Spindle DC line to monitor router load. Monitoring load can improve the quality of cut parts. 

I am certain the function was introduced to aid in monitoring current to the laser drivers. But I think it can be used for much more given that a clever person can derive a lot of data from these chips. 

To be clear I am not requesting any changes to interface. Only serial output, maybe a clever way to view the data as a sub menu of the dot matrix type screens. 

### Additional context

I do believe the request, at least right now is of very low priority. I do not expect to see this request implemented. But I am hopeful that some one might see the value and work in a way to implement some expanded functionality for multiple I2C voltmeter/Ammeter channels.

toxicpsion, 2024-05-22:
I have working code for this, but need comments before formatting a PR.

why are we pulling in Wire and INA226 for 5 i2c reads in ammeter.cpp? that locks us to SDA/SCL.

i propose to pull SlowSoftI2CMaster instead, to allow remapping pins, as the MKS GEN L v1 i'm tinkering on for this laser doesn't expose SDA/SCL. I feel it would probably be more useful that way?

rewriting the existing INA226 code to use the same is fairly trivial, but i would like some feedback before cooking something up.

rflulling, 2024-05-30:
Thank you both, @toxicpsion and @thisiskeithb . 
Some one (toxicpsion) had just been asking about INA226 on Discord feature requests. I remembered asking about this long ago when I saw the addition of a single chip single read I think for laser use. -I hoped then for additional option to use several as I2C may permit, and or a INA3221.  However, I had forgotten entirely that I had started this FR. I was thinking the entire conversation had taken place on Discord, and maybe part of it did. I never would have guessed this would be dug up and referenced. 

Today I was rummaging around my profile in GIThub and noticed this FR tied to my profile. -Hmm whats this? 
Well Color me surprised you have both been working on this. I honestly would have not been surprised if a FR from 2021 had been closed with no activity, heck I'd forgotten about it. 

I look forward to seeing what you come up with and you incorporate!

-My best hope was to at least get an option to output to serial where even if the display and software didn't use the data, I might be able to monitor manually and maybe even write a script to also follow serial and log the data with time stamp. 

Now if nifty toys are added to the firmware, even better!
-Other than logging, a clever function might see the data used similar to how we use PID auto tune to monitor temp of a hot end. Except rather than arbitrary, we can know whats going on by directly monitoring the Voltage and Current, and possibly even dynamically adjusting the current. For TMC motors, the code allows us to adjust current, or reduce it. For Fans and Hot ends we can know if there is a short circuit or break in the wire and either set of a notice or go into thermal shutdown. For Spindle or laser we can monitor loading. For nearly any part in the system we can monitor operational hours and set off a notice if the part has exceeded a value for recommended replacement or advanced warning of failure. -I will suggest that if any internal logging is done, it REQUIRE an SD card to load and store the data too.