MarlinFirmware/Marlin #23207: [BUG] Line error on TFT35 E3 v3 display using touchscreen mode while connected to SKR Mini E3 v2

silverdragonhawk, 2021-11-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I compiled the latest marlin 2.0 bugfix under both build environments STM32F103RC_btt_USB and STM32F103RC_btt_USB_maple and I get the same result, after turning on my printer it still says no printer connected, and I repeatedly get a error pop up saying "Line Number is not last line Number+1, Last Line: 0"

This happens regardless if I have octoprint connected and running or not.

### Bug Timeline

This started about 3 days ago

### Expected behavior

I expected my touchscreen display to connect to the printer and work

### Actual behavior

After turning on my printer it still says no printer connected, and I repeatedly get a error pop up saying "Line Number is not last line Number+1, Last Line: 0"

### Steps to Reproduce

1) compile marlin 2.0 bugfix under STM32F103RC_btt_USB and/or STM32F103RC_btt_USB_maple
2) flash firmware.
3) start printer with display in touchscreen mode

### Version of Marlin Firmware

Marlin 2.0 bugfix

### Printer model

Ender 3 Pro

### Electronics

SKR Mini E3 v2, TFT35 E3 v3, BL Touch v3.1, and Octoprint (windows 10)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[firmware files.zip](https://github.com/MarlinFirmware/Marlin/files/7610999/firmware.files.zip)

gmoney121, 2021-11-28:
Im not of any help but I am having the same issue. im using a tft 70 on a skr e3 turbo and have the same exact isssue. I tried compiling a fresh firmware unmodified but got the same issue so it must be a bug.

jstaylor11, 2021-12-07:
I am experiencing the exact same problem (same hardware as well: SKR Mini E3 v2, TFT35 E3 v3, BL Touch v3.1).

radek8, 2021-12-11:
Update FW TFT to the latest versions.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

radek8, 2021-12-11:
Your problem is not related to Marlin but TFT

silverdragonhawk, 2021-12-12:
Thank you, I'll close this then.

gmoney121, 2021-12-12:
its not TFT firmware the issue came up after updating marlin. I even tried updating the TFT again even though I did it before and it still gives the line error. even tried with fresh firmware on both and same thing.

silverdragonhawk, 2021-12-12:
hmm, I'll test this later tonight or tomorrow and report back.

radek8, 2021-12-12:
see
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2265

silverdragonhawk, 2021-12-12:
Thanks @radek8 . I tested and can confirm it is fixed. Greatly appreciate you pointing me in the right direction! I'm still getting the hang of this github thing, lol. This issue will now be closed.

closed 2021-12-12 by silverdragonhawk