MarlinFirmware/Marlin #23209: [BUG] temp error while bed levelling

sklero001, 2021-11-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I installed the version of marlin 2.0.9 on a mega2560 rev A + and I have a problem with the temperature ... when in my printer I send the command for leveling the plane (with the bltouch) every time the pin of the bltouch it goes down, the temperature sensors stop working ... Using a multi-point leveling, it then gives me a minimum temperature error of E. If I don't use the self-leveling of the hob, everything works perfectly. So I assume it's an interference between the sensors and the bltouch. How can I fix it ??

### Bug Timeline

new bug

### Expected behavior

complete auto bed leveling

### Actual behavior

error_min_temp E

### Steps to Reproduce

1. lanch auto bed levelling on more points

### Version of Marlin Firmware

2.0.9.2

### Printer model

Geeetech i3 pro b

### Electronics

stock

### Add-ons

BLtouch V3

### Bed Leveling

ABL Bilinear mesh

### Additional information & file uploads

zip with configuration.h and confiiguration_adv.h
https://drive.google.com/drive/folders/1jK3_ZbKqKP0kev1TiaHyIUUUTWZFZ9PW?usp=sharing

robbycandra, 2021-11-29:
I think it is a hardware issue.

sklero001, 2021-11-29:
Hi and thanks for your support...
with Marlin 1.0.8 it work fine!! but with Marlon 2.0.9 nope...

Tannoo, 2021-12-01:
I don't have that issue with my printer. I don't believe this is a software issue.

This is usually an issue with the thermistor or it's wiring.

closed 2022-01-09 by thisiskeithb