MarlinFirmware/Marlin #23210: [BUG] Compile Error STM32 in 2.0.9.2 and bugfix 2.x

Papipapito, 2021-11-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compile show error:

Compiling .pio\build\STM32F103VC_ZM3E4_USB\FrameworkArduino\abi.cpp.o
C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp: In function 'void tone(uint8_t, unsigned int, long unsigned int)':
<command-line>: error: invalid conversion from 'int' to 'TIM_TypeDef*' [-fpermissive]
C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp:111:35: note: in expansion of macro 'TIMER_TONE'
  111 |     TimerTone = new HardwareTimer(TIMER_TONE);
      |                                   ^~~~~~~~~~
In file included from C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:45,
                 from C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino\wiring.h:41,
                 from C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino\Arduino.h:36,
                 from C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp:22:
C:\Users\alber\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareTimer.h:101:32: note:   initializing argument 1 of 'HardwareTimer::HardwareTimer(TIM_TypeDef*)'
  101 |     HardwareTimer(TIM_TypeDef *instance);
      |                   ~~~~~~~~~~~~~^~~~~~~~
*** [.pio\build\STM32F103VC_ZM3E4_USB\FrameworkArduino\Tone.cpp.o] Error 1

### Bug Timeline

new

### Expected behavior

Compile good

### Steps to Reproduce

Download marlin fresh version

Change Board, and drivers , and PlatformIO file and compile

### Version of Marlin Firmware

2.0.9.2

### Printer model

Zonestar Z9M4

### Electronics

BOARD_ZONESTAR_ZM3E4

Papipapito, 2021-11-27:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7612144/Marlin.zip)

ellensp, 2021-11-28:
In ini/stm32f1.ini  change -DTIMER_TONE=1 to -DTIMER_TONE=TIM1

Ie find
```
[env:STM32F103VC_ZM3E4_USB]
platform             = ${ZONESTAR_ZM3E.platform}
extends              = ZONESTAR_ZM3E
board                = genericSTM32F103VC
board_build.variant  = MARLIN_F103Vx
build_flags          = ${ZONESTAR_ZM3E.build_flags} -DTIMER_TONE=1
```

and change it to 
```
[env:STM32F103VC_ZM3E4_USB]
platform             = ${ZONESTAR_ZM3E.platform}
extends              = ZONESTAR_ZM3E
board                = genericSTM32F103VC
board_build.variant  = MARLIN_F103Vx
build_flags          = ${ZONESTAR_ZM3E.build_flags} -DTIMER_TONE=TIM1
```

Papipapito, 2021-11-28:
Hello

Works!!

With this change , compile correctly

Thanks

ellensp, 2021-12-03:
This has now been added to bugfix. closing.

closed 2021-12-03 by ellensp