MarlinFirmware/Marlin #23213: Compile Error for Marlin bugfix-2.0.x with Creality V4.2.7 & Ender 5

zacharyalan09, 2021-11-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I receive this error when trying to compile Marlin with attached files.

I see the same issue with the example version config files for Ender-5 Creality V427.

[Marlin-Clean.zip](https://github.com/MarlinFirmware/Marlin/files/7612675/Marlin-Clean.zip)

Marlin\src\HAL\STM32\fast_pwm.cpp: In function 'void set_pwm_frequency(pin_t, int)':
Marlin\src\HAL\STM32\fast_pwm.cpp:74:36: error: 'Instance' was not declared in this scope
   74 |   uint32_t index = get_timer_index(Instance);
      |                                    ^~~~~~~~
Marlin\src\HAL\STM32\fast_pwm.cpp:77:16: error: 'TEMP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                ^~~~~~~~~~~~~~
      |                TIMER_NUM
Marlin\src\HAL\STM32\fast_pwm.cpp:77:43: error: 'STEP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                                           ^~~~~~~~~~~~~~
      |                                           TIMER_NUM
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32\tft\tft_ltdc.cpp.o
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32\fast_pwm.cpp.o] Error 1
=========================================================================== [FAILED] Took 7.65 seconds ===========================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:07.646
====================================================================== 1 failed, 0 succeeded in 00:00:07.646 ======================================================================The terminal process "C:\Users\Zach\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

### Bug Timeline

This is the first Marlin version I've tried to compile.

### Version of Marlin Firmware

2.0.x Bugfix

### Printer model

Ender 5

### Electronics

V4.2.7, BLTouch,

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

thisiskeithb, 2021-11-28:
> This is the first Marlin version I've tried to compile.

What luck. This was changed just two hours ago: https://github.com/MarlinFirmware/Marlin/commit/0539e870de30877a38c78d29a304a97a6068cc80

It should be fixed in https://github.com/MarlinFirmware/Marlin/pull/23187

zacharyalan09, 2021-11-28:
Thanks Keith - Is there a way to pull the files down before they're published in the newest release? Or will they be published overnight with the latest bugfix files for Marlin?

tpruvot, 2021-11-28:
you can also revert the last commit of bugfix

dcs3473, 2021-11-28:
This is still broken for me also.

thisiskeithb, 2021-11-28:
> This is still broken for me also.

Yes, this issue is marked as `Bug: Confirmed!` and will affect everyone trying to compile the latest `bugfix-2.0.x` until #23187 is merged or bug fixed directly.

thisiskeithb, 2021-11-29:
@thinkyhead Can you rename this issue so it's easier to find? We're going to be slammed closing issues until https://github.com/MarlinFirmware/Marlin/pull/23187 is merged.

Sonicboom247, 2021-11-29:
Thanks! just tested with the new file and now able to compile for Artillery Ruby board

KillerSquid, 2021-11-30:
I'm still getting this error when trying to compile

`Marlin/src/HAL/STM32/fast_pwm.cpp: In function 'void set_pwm_frequency(pin_t, int)':
Marlin/src/HAL/STM32/fast_pwm.cpp:74:36: error: 'Instance' was not declared in this scope
   74 |   uint32_t index = get_timer_index(Instance);
      |                                    ^~~~~~~~
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/fast_pwm.cpp.o] Error 1`

walterlootk, 2021-11-30:
So lucky. I just downloaded this version, did all the necessary changes for DELTA, and compiled. The exact three error messages came out.

Building in release mode
Compiling .pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\fast_pwm.cpp.o
Compiling .pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\tft\tft_ltdc.cpp.o
Compiling .pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\tft\tft_spi.cpp.o
Compiling .pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\tft\xpt2046.cpp.o
Marlin\src\HAL\STM32\fast_pwm.cpp: In function 'void set_pwm_frequency(pin_t, int)':
Marlin\src\HAL\STM32\fast_pwm.cpp:74:36: error: 'Instance' was not declared in this scope
   74 |   uint32_t index = get_timer_index(Instance);
      |                                    ^~~~~~~~
Marlin\src\HAL\STM32\fast_pwm.cpp:77:16: error: 'TEMP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                ^~~~~~~~~~~~~~
      |                TIMER_NUM
Marlin\src\HAL\STM32\fast_pwm.cpp:77:43: error: 'STEP_TIMER_NUM' was not declared in this scope; did you mean 'STEP_TIMER'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                                           ^~~~~~~~~~~~~~
      |                                           STEP_TIMER
*** [.pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\fast_pwm.cpp.o] Error 1

ellensp, 2021-11-30:
ATTENTION ALL NEW POSTERS

This has not been fixed yet, Is waiting on https://github.com/MarlinFirmware/Marlin/pull/23187

Be patient.

Sonicboom247, 2021-11-30:
for those impatient ones like me, you can copy the code from here 
https://github.com/MarlinFirmware/Marlin/blob/da7d90c03cb96a92cb978836e140e406a48535ee/Marlin/src/HAL/STM32/fast_pwm.cpp and here https://github.com/MarlinFirmware/Marlin/blob/da7d90c03cb96a92cb978836e140e406a48535ee/Marlin/src/HAL/STM32/timers.h

into your respective files

zacharyalan09, 2021-11-30:
I copied in the linked code and was able to compile with the new fast_pwm.cpp and timers.h files on a Creality V4.2.7 board.

thisiskeithb, 2021-11-30:
Fixed in c09641c.

closed 2021-11-30 by thisiskeithb