MarlinFirmware/Marlin #23214: [BUG] GTM32 REV B board - wrong pin configuration.

agressiva, 2021-11-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I tested my m201 printer with GTM32 REV B board and the extruder take on fire.

After some checks i see that all fets that control the bed and extrudr hot ends are aways ON.

After that i tested x and y motors and it not move.

the extruders sensors are wrong too.

Does anyone know where i can find the schematic os GTM32 REV B board ?

### Bug Timeline

new bug

### Steps to Reproduce

POWER On the board and see what heapens !

### Version of Marlin Firmware

2.0.9.2

### Printer model

geeetech m201

### Electronics

stock electronics

ellensp, 2021-11-28:
this is met to be it https://www.geeetech.com/forum/download/file.php?id=7140

agressiva, 2021-11-29:
I do the reverse engineering and have almost all pins configuration for this board.

with this pinout all motors and heaters are working, but
I have some strange things on when moving motors.

With parameters (accel and feed rate used on my m201 8bit board) the motors turn super fast losing almost all steps.

I need reduced the acceleration to 2 and feedrate to 3 so i can made motors turn
but it turn very strange... 

bellow are the first draft pinout for this board
_________________________________________________________

GEEETECH GTM32 REV_B
REVERSE ENGINEERING BY EDUARDO LAVRATTI

X dir       - 67  PA8
X step      - 77  PA15
X ena       - 84  PD3

Y dir       - 96  PB9
Y step      - 95  PB8
Y ena       - 87  PD6

Z dir       - 47  PB10
Z step      - 37  PB2
Z ena       - 59  PD12

E0 dir      - 58  PD11
E0 step     - 89  PB3
E0 ena      - 60  PD13

E1 dir      - 93  PB7
E1 step     - 90  PB4
E1 ena      - 07  PC13

E2 dir      - 36  PB1
E2 step     - 33  PC4
E2 ena      - 51  PB12

SERIAL COM  - 79  PC11
SERIAL COM  - 78  PC10

X MIN       - 98  PE1
X MAX       - 97  PE0

Y MIN       - 2   PE3
Y MAX       - 1   PE2

Z MIN       - 4   PE5
Z MAX       - 3   PE4

T_BED       - 18  PC3
T_E2        - 17  PC2
T_E1        - 16  PC1
T_E0        - 15  PC0

FAN_0       - 92  PB6
FAN_1       - 70  PA11
FAN_2       - 35  PB0

HETA_0      - 31  PA6
HEAT_1      - 32  PA7
HEAT_2      - 24  PA1
H_BED       - 23  PA0

beep        - 91  PB5

display pins - need confirmation
64  PC7    ??? LCD_PINS_ENABLE
71  PA12   ??? LCD_PINS_RS

??  ???    ??? LCD_PINS_D4
85  PD4  - ??? PCD_PINS_D5
86  PD5  - ??? LCD_PINS_D6
88  PD7  - ??? LCD_PINS_D7

81  PD0  - ???
91  PB5  - ???
63  PC6  - ???

ellensp, 2021-12-01:
Looks like the pins file was incorrectly copied from pins_GTM32_PRO_VB.h
The initial commit even says GTM32 Pro VB https://github.com/MarlinFirmware/Marlin/pull/16020/files

ellensp, 2021-12-01:
@Vertabreak can you take a look at this?
are there two versions of the board?

Vertabreak, 2021-12-01:
geeetech is notorious for making undocumented revisions to hardware, this being said the pins was tested on a board from 2017 ish.

closed 2022-03-21 by github-actions[bot]