MarlinFirmware/Marlin #23219: [BUG] Bilinear Leveling Grid inverted on Y axis

SiegelDaniel, 2021-11-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am running the latest Marlin tag and already tried bugfix, however it still occurs:

I have a slightly concave bed and I cannot get any print to stick to the bed at the center. However, whenever I print on the right or left hand side everything works fine. 

My grid looks as following:

`      0      1      2      3      4
 0 +0.024 -0.030 +0.059 +0.094 +0.096
 1 +0.072 +0.094 +0.113 +0.179 +0.082
 2 +0.033 +0.230 -0.014 +0.275 +0.180
 3 +0.002 +0.116 +0.164 +0.305 +0.200
 4 -0.010 -0.132 +0.006 +0.112 +0.073`

What may be the error behind this? Line 0 should be at Line 4 position, Line 1 should be at Line 3 position.
I suspect this is why my printer is moving the nozzle up on the concave part of the bed.

### Bug Timeline

New

### Version of Marlin Firmware

2.0.9.2

### Printer model

Creality CR-10 mini

### Electronics

MKS SGEN L with TMC2208 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7614300/Marlin.zip)

sjasonsmith, 2021-11-28:
I believe the rows are reported in the order they are probed, so it looks upside down when looking at the table as if you were looking at the bed. In this case the first point reported is the point closest to your origin (0,0) point.

sjasonsmith, 2021-11-28:
I am hoping somebody else will drop in to either confirm or debunk my assertion above.

robbycandra, 2021-11-29:
Nothing wrong with this. It seems inverted because Cartesian Coordinate puts Y=0 at the bottom.

borland1, 2021-11-29:
Seems to be inverted to me.  Most concerning is that data is presented without proper labels, so in an ambiguous fashion.  Unless 'X' and 'Y' are labeled, there is no way to know if data is displayed as inverted.   
```
> G29 S0
Num X,Y: 3,3
Z offset: 0
Measured points:
       0      1       2
0 +0.011 -0.020  -0.026
1 +0.017 +0.002  -0.019
2 +0.022 -0.030  -0.013
```
Suggest that `print_2d_array()` in file `bedlevel.cpp` be modified to define X, Y labels, so that when it is called by `report_mesh()`, the data will be displayed in an unambiguous way.

wheelyjoe, 2022-01-01:
I seem to be havign the same issue, though around the X axis, ie left-to-right inversion. It also seems to be compensating using the inverted grid, making my lows lower and my highs higher. 

I have deliberately set the RIGHT side of my bed to be higher than the left, and this is the G29 output: 

`Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +1.434 +0.294 -0.857
Recv:  1 +1.396 +0.240 -0.959
Recv:  2 +1.342 +0.118 -1.081`

thinkyhead, 2022-01-05:
Inversion would be a pretty glaring bug, yes? Something else is at issue. Try the `X_AXIS_TWIST_COMPENSATION` feature and see if it reveals anything interesting.

closed 2022-03-18 by github-actions[bot]