MarlinFirmware/Marlin #23220: [BUG] Compile error, Marlin-bugfix-2.0.x BTT SKR Mini E3 v2.0/Ender 3

Snargola, 2021-11-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm attempting to compile the latest Marlin-bugfix-2.0.x (downloaded 5 mins ago) and am running into the following error:
Marlin\src\HAL\STM32\fast_pwm.cpp: In function 'void set_pwm_frequency(pin_t, int)':
Marlin\src\HAL\STM32\fast_pwm.cpp:74:36: error: 'Instance' was not declared in this scope
   74 |   uint32_t index = get_timer_index(Instance);
      |                                    ^~~~~~~~
Marlin\src\HAL\STM32\fast_pwm.cpp:77:16: error: 'TEMP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                ^~~~~~~~~~~~~~
      |                TIMER_NUM
Marlin\src\HAL\STM32\fast_pwm.cpp:77:43: error: 'STEP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                                           ^~~~~~~~~~~~~~
      |                                           TIMER_NUM
*** [.pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\fast_pwm.cpp.o] Error 1
========================================================== [FAILED] Took 83.35 seconds ==========================================================
Environment          Status    Duration
-------------------  --------  ------------
STM32F103RC_btt_USB  FAILED    00:01:23.346
===================================================== 1 failed, 0 succeeded in 00:01:23.346 ===================================================== 

I will attach my configuration.h and configuration_adv.h here. I have compiled Marlin numerous times for this, and other, machines and have never had this error before. I am at a loss.
[files.zip](https://github.com/MarlinFirmware/Marlin/files/7614429/files.zip)

### Bug Timeline

New too me. It started about 5 minutes ago

### Expected behavior

I expected it to compile error free.

### Actual behavior

It did not compile error free.

### Steps to Reproduce

1) Download the latest bugfix Marlin
2) Attempt to compile
3) ??
4) Profit!!

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

Creality Ender 3

### Electronics

BTT SKR Mini E3 v2

### Additional information & file uploads

Thank you for your assistance.

thisiskeithb, 2021-11-28:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23213

closed 2021-11-28 by thisiskeithb

Snargola, 2021-11-28:
I don't know if it matters but it's a different board/machine.

ellensp, 2021-11-29:
@Snargola not in this case both same HAL\STM32