MarlinFirmware/Marlin #23221: [BUG] Compile error with BIGTREE_SKR_PRO board regarding undeclared variables within scope

TeknoloGenie, 2021-11-28:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Im building Marlin 2.0.9.2 with board BIGTREE_SKR_PRO and im getting the following build error. Im using VSCode with the PlatformIO IDE extension and here is the error,

```

In file included from Marlin\src\HAL\STM32\tft\xpt2046.cpp:28:
Marlin\src\HAL\STM32\tft\xpt2046.h:35:4: error: #error "TOUCH_MISO_PIN is not defined."
   35 |   #error "TOUCH_MISO_PIN is not defined."
      |    ^~~~~
In file included from C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:4,
                 from C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22,
                 from C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
                 from C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:44,
                 from C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
                 from C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
                 from C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:36,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32\tft\xpt2046.cpp:24:
Marlin\src\HAL\STM32\tft\xpt2046.h: In static member function 'static void XPT2046::DataTransferBegin()':
Marlin\src\HAL\STM32\tft\xpt2046.h:72:94: error: 'TOUCH_CS_PIN' was not declared in this scope; did you mean 'TOUCH_INT_PIN'?
   72 |   static inline void DataTransferBegin() { if (SPIx.Instance) { HAL_SPI_Init(&SPIx); } WRITE(TOUCH_CS_PIN, LOW); };
      |                                                                                              ^~~~~~~~~~~~
C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:58:49: note: in expansion of macro 'digitalPinToPinName'
   58 |   #define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->BSRR = _BV32(STM_PIN(digitalPinToPinName(IO)) + ((V) ? 0 : 16)))
      |                                                 ^~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin\src\HAL\STM32\tft\xpt2046.h:72:88: note: in expansion of macro 'WRITE'
   72 |   static inline void DataTransferBegin() { if (SPIx.Instance) { HAL_SPI_Init(&SPIx); } WRITE(TOUCH_CS_PIN, LOW); };
      |                                                                                        ^~~~~
Marlin\src\HAL\STM32\tft\xpt2046.h: In static member function 'static void XPT2046::DataTransferEnd()':
Marlin\src\HAL\STM32\tft\xpt2046.h:73:48: error: 'TOUCH_CS_PIN' was not declared in this scope; did you mean 'TOUCH_INT_PIN'?
   73 |   static inline void DataTransferEnd() { WRITE(TOUCH_CS_PIN, HIGH); };
      |                                                ^~~~~~~~~~~~
C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
  129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
      |                                  ^
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:58:49: note: in expansion of macro 'digitalPinToPinName'
   58 |   #define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->BSRR = _BV32(STM_PIN(digitalPinToPinName(IO)) + ((V) ? 0 : 16)))
      |                                                 ^~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:69:33: note: in expansion of macro '_WRITE'
   69 | #define WRITE(IO,V)             _WRITE(IO,V)
      |                                 ^~~~~~
Marlin\src\HAL\STM32\tft\xpt2046.h:73:42: note: in expansion of macro 'WRITE'
   73 |   static inline void DataTransferEnd() { WRITE(TOUCH_CS_PIN, HIGH); };
      |                                          ^~~~~
In file included from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/HAL.h:31,
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32\tft\xpt2046.cpp:24:
Marlin\src\HAL\STM32\tft\xpt2046.cpp: In static member function 'static void XPT2046::Init()':
Marlin\src\HAL\STM32\tft\xpt2046.cpp:38:13: error: 'TOUCH_CS_PIN' was not declared in this scope; did you mean 'TOUCH_INT_PIN'?
   38 |   OUT_WRITE(TOUCH_CS_PIN, HIGH);
      |             ^~~~~~~~~~~~
Marlin\src\HAL\STM32\tft\../../../inc/../HAL/./STM32/fastio.h:66:41: note: in definition of macro '_SET_OUTPUT'
   66 | #define _SET_OUTPUT(IO)         pinMode(IO, OUTPUT)                               //!< Output Push Pull Mode & GPIO_NOPULL
      |                                         ^~
Marlin\src\HAL\STM32\tft\xpt2046.cpp:38:3: note: in expansion of macro 'OUT_WRITE'
[... 85 lines cut ...]
                 from Marlin\src\HAL\STM32\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32\tft\tft_spi.cpp:24:
Marlin\src\HAL\STM32\tft\tft_spi.cpp:40:75: error: 'TFT_SCK_PIN' was not declared in this scope; did you mean 'SD_SCK_PIN'?
   40 |   if ((spiInstance = (SPI_TypeDef *)pinmap_peripheral(digitalPinToPinName(TFT_SCK_PIN),  PinMap_SPI_SCLK)) == NP) return;
      |                                                                           ^~~~~~~~~~~
C:\Users\NodeDad\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:249:49: note: in definition of macro 'digitalPinToPinName'
  249 | #define digitalPinToPinName(p)      (((uint32_t)p < NUM_DIGITAL_PINS) ? digitalPin[p] : NC)
      |                                                 ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\tft\xpt2046.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\tft\tft_spi.cpp.o] Error 1
========================================================================================== [FAILED] Took 12.32 seconds ==========================================================================================
Environment      Status    Duration
---------------  --------  ------------
BIGTREE_SKR_PRO  FAILED    00:00:12.320
===================================================================================== 1 failed, 0 succeeded in 00:00:12.320 =====================================================================================The terminal process "C:\Users\NodeDad\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

### Version of Marlin Firmware

2.0.9.2

TeknoloGenie, 2021-11-28:
updated post with bugfix branch build

thisiskeithb, 2021-11-28:
> updated post with bugfix branch build

> `Marlin\src\HAL\STM32\fast_pwm.cpp:77:16: error: 'TEMP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?`

That bug has been reported in #23213 and will be fixed in #23187

The SKR Pro's pins file likely needs to be updated for whatever TFT you have enabled. As for your build error, please attach your configs since we don't know what hardware you're using.

viper93458, 2021-11-28:
Same error on my SKR 1.1 that's running my delta with a 12864 reprap discount full graphic display:

Marlin\src\HAL\STM32\fast_pwm.cpp:77:16: error: 'TEMP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                ^~~~~~~~~~~~~~
      |                TIMER_NUM
Marlin\src\HAL\STM32\fast_pwm.cpp:77:43: error: 'STEP_TIMER_NUM' was not declared in this scope; did you mean 'TIMER_NUM'?
   77 |   if (index == TEMP_TIMER_NUM || index == STEP_TIMER_NUM
      |                                           ^~~~~~~~~~~~~~
      |                                           TIMER_NUM
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\fast_pwm.cpp.o] Error 1

-William

thisiskeithb, 2021-11-28:
> Same error on my SKR 1.1 that's running my delta with a 12864 reprap discount full graphic display:

Like I said above, That bug has been reported in #23213 and will be fixed in #23187. It's unrelated to your current issue.

Please attach the SKR Pro configs that you opened this issue for.

viper93458, 2021-11-28:
I didn't intend to hijack just wanted to post a confirmation from another user. That's all. BTW: Reverting the "Rename HAL Timer elements" commit resolved it for me without waiting on a new PR. :)

TeknoloGenie, 2021-11-29:
Im building my own custom printer with 1500Z 600Y and 900X axis, i have not done any config files, just stared a tutorial from this youtube video youtube.com/watch?v=mKMNtfyX36A all i did thus far is download the marlin firmware and change the platformio.ini environment and configuration.h files motherboard to the pro. Would you like me to attach my entire Configuration.h file?

TeknoloGenie, 2021-11-29:
OMG! Thank you so much SUCCESS on my build after pulling in the changes from that issues. THANK YOU THANK YOU!

thisiskeithb, 2021-11-29:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23213

thisiskeithb, 2021-11-29:
> OMG! Thank you so much SUCCESS on my build after pulling in the changes from that issues. THANK YOU THANK YOU!

I don't see how that fixed the pin errors, but glad you got it working.

closed 2021-11-29 by thisiskeithb