MarlinFirmware/Marlin #23223: [BUG] Chitu V5 board with 3.5" TFT, screen is black (Tronxy XY2-PRO)

DByhlin, 2021-11-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling with PlatformIO (chitu_v5_gpio_init), BOARD_CHITU3D_V5 and TFT_TRONXY_X5SA, when running the update.cbd-file i get the usual beeps, but nothing happens on the screen. It just stays black. When compiling with the same parameters on this release: https://github.com/rhapsodyv/Marlin, the screen works. Tried the GENERIC TFT with same results.

### Bug Timeline

New

### Expected behavior

I expected to get the Classic TFT UI on the screen.

### Actual behavior

The screen stays unlit and black.

### Steps to Reproduce

Compile using the chitu_v5 parameters
Update with the update-file
Start printer

### Version of Marlin Firmware

2.0.9.2

### Printer model

Tronxy XY2-Pro

### Electronics

Stock

### Add-ons

Disabled bed leveling sensor

### Your Slicer

Prusa Slicer

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7614610/configs.zip)

ellensp, 2021-11-29:
This will be the same issue as https://github.com/MarlinFirmware/Marlin/issues/23198

DByhlin, 2021-11-29:
@ellensp thank you, this solved the issue.

closed 2021-11-29 by DByhlin