MarlinFirmware/Marlin #23227: Quad Z Quad Endstop Homing Issue

TheSWilde, 2021-11-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Build a 3D-printer with quad Z stepper and quad Z stop. Each stepper has its own driver (TMC2209 UART) on a BTT Octopus 1.1.
Whenever i home the Z-axis the gantry drives down until touch each Z-stop. Than it drives up and slowly back down to the Z-Stops.
Than errormessage "ERROR:Pinter haltet. kill() called!" appear and i have to reset the printer.

If i configure Marlin to just use one oft the 4 endstops, everything works fine. 

### Bug Timeline

Old issue  (#17785?) without solution

### Expected behavior

Z-gantry drives down to Z-Stops, a bit up, than slowly down and tadaa: Z is at 0 and printer ready.

### Actual behavior

Z homing touch up endstops twice and printer gets stopped with errormessage  "ERROR:Pinter haltet. kill() called!" 

### Steps to Reproduce

The error appears as soon as i homing Z. Everytime. 

### Version of Marlin Firmware

2.0.9.2

### Printer model

DIY

### Electronics

BTT Octopus V1.1 + 7x TMC2209 UART-Mode

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Errorlog.txt](https://github.com/MarlinFirmware/Marlin/files/7619495/Errorlog.txt)
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7619497/configs.zip)

ellensp, 2021-11-30:
What does M119 say with no endstops triggered?

ellensp, 2021-11-30:
if you disable VALIDATE_HOMING_ENDSTOPS it doesn't ERROR
Still looking into why

ellensp, 2021-11-30:
also fails on 3 Z steppers.

ellensp, 2021-11-30:
hit_state is definitely going to 0 on 3rd Z stepper home, this triggers the error.  but I cant find out where/what is going on before that.

TheSWilde, 2021-11-30:
I also tried disable VALIDATE_HOMING_ENDSTOPS for testing.
I was confused, because: X and Y drives straight to their stops and are zeroed without a second run (as expected).
Z drives down until every stop is triggered, drives a bit up and does a second run until Z-stops are triggered again (without any error)
Shouldn't the second run be disabled?

thinkyhead, 2021-12-07:
Does the machine still go to KILL state if you disable `USE_WATCHDOG` ?

TheSWilde, 2021-12-07:
The error occurs even with Watchdog disabled

tu77, 2024-01-05:
**same Problem here!**

I upgraded from 2 to 3 Z axis and got the same issue!
in the end I downgraded to 2 Axis again.. a bit annoying!

latest marlin
BTT Octopus 1.1 
Core XY

Ted-Printable, 2024-09-06:
https://github.com/MarlinFirmware/Marlin/issues/26717  has a lot information.

To summarize:

Running Marlin2.1.2.4 (bugfix version late August had same results)  on a BTT Octopus V1.1 (STM32F446ZE) with TMC2209 drivers.
The machine has FOUR - Z-steppers, Z, Z2, Z3 and Z4 with four Z endstops.

Homing while using only one endstop ends normally with normal results, accept of course no stepper alignment.
Homing with multiple endstops without "validation" , meaning disabling VALIDATE_HOMING_ENDSTOPS, terminate normal and correct. All four steppers are aligned and homed, with ONE bump only.

However, using the Validation feature (which is a great feature) results in "HALT CALLED" after the second, or validating bump.
It appears that at least part of the issue is the 4 endstop activations occurring at almost the same moment. Marlin calls KILL because one or more endstops were "not seen to be activated". It seems that four endstops occurring the same moment cannot be correctly detected by Marlin interrupts, which is probably a physical limitation.

A possible solution would be to have Marlin accept that the "validation bump" is good as long as it received at least one endstop signal. This would work as the first bump still works normal to align the four axes perfectly and the validation is only to get a good zero point with a slower bump. 
A possible work around would be to enable users to specify individual axis to validate. That way at least X and Y can be validated properly.
All this was deduced from results after  disabling the kill call in "endstops.ccp". In file \Marlin\src\module\endstops.cpp on lines 453 to 459 is the statement that kills the printer if any endstop is not triggered. By disabling this, the homing ends normal, but with no error checking on any axis, X,Y and Z, which is not good at all.

This need to be altered to ignore if any of Z, Z2, Z3, Z4 is not triggered. It should be ok if only one of the four is triggered on the second bump only. (I think the second bump is a problem as it triggers all four endstops almost simultaneously, resulting in one or more triggers being missed by Marlin.)

Someone with better programming skills could implement that or hopefully a better solution.
There is still the mystery of multiple homing calls that was first noticed by "ellensp"  (https://github.com/ellensp)
EllenSP is right by saying more debugging is needed as the bug is still not isolated completely.

All help and assistance is greatly appreciated.
Thanks you.

thinkyhead, 2025-10-02:
I don't know if you saw the solution proposed by @sorgelig over at #27983. It was posted there and one of our helpful triage volunteers pointed out that it was a "duplicate issue" and invited the author to come *all the way over here* and get involved in helping out. But they were not in the mood so they just flew off in a fit of pique.

I have been consumed with the motion system, the march towards a 2.1.3 release, and I have been taking a much-needed mental health break over the summer, so I am only now reviewing these issues on the eve of the next release. The solution seems clean and elegant but it needs to be evaluated and tested on real hardware.

As it happens, I have some parrot DNA so I am happy to post their proposal here:

```diff
diff --git a/Marlin/src/module/motion.cpp b/Marlin/src/module/motion.cpp
index 0ff5db4..5130e8b 100644
--- a/Marlin/src/module/motion.cpp
+++ b/Marlin/src/module/motion.cpp
@@ -2176,25 +2176,25 @@ void prepare_line_to_destination() {
             if (pos_dir) {
               // normalize adj to smallest value and do the first move
               (*lock[0])(true);
-              do_homing_move(axis, adj[1] - adj[0]);
+              if(adj[1] - adj[0]) do_homing_move(axis, adj[1] - adj[0]);
               // lock the second stepper for the final correction
               (*lock[1])(true);
-              do_homing_move(axis, adj[2] - adj[1]);
+              if(adj[2] - adj[1]) do_homing_move(axis, adj[2] - adj[1]);
               #if NUM_Z_STEPPERS >= 4
                 // lock the third stepper for the final correction
                 (*lock[2])(true);
-                do_homing_move(axis, adj[3] - adj[2]);
+                if(adj[3] - adj[2]) do_homing_move(axis, adj[3] - adj[2]);
               #endif
             }
             else {
               #if NUM_Z_STEPPERS >= 4
                 (*lock[3])(true);
-                do_homing_move(axis, adj[2] - adj[3]);
+                if(adj[2] - adj[3]) do_homing_move(axis, adj[2] - adj[3]);
               #endif
               (*lock[2])(true);
-              do_homing_move(axis, adj[1] - adj[2]);
+              if(adj[1] - adj[2]) do_homing_move(axis, adj[1] - adj[2]);
               (*lock[1])(true);
-              do_homing_move(axis, adj[0] - adj[1]);
+              if(adj[0] - adj[1]) do_homing_move(axis, adj[0] - adj[1]);
             }
```

closed 2025-10-03 by thinkyhead

thinkyhead, 2025-10-03:
Ah, I see! I recently got a note from someone using our CAN bus support, reporting that it was working very well for them.

So we just need some nerd who is knowledgeable about the protocols of the EBB to come along and add those special commands to the code. I could research it myself but my current priority is to clear up lingering bugs and get this 2.1.3 release out the door.

BigTreeTech is still providing us some funding, but has not been actively participating in development for many months. So we may have to scrape Klipper code to figure out those protocols.

commit 3b149c35 (2025-10-03): Skip zero length homing_move (#28090)

Fixes #23227