MarlinFirmware/Marlin #23236: [BUG] (E3D Kraken extruder correctly configured powers more than one nozzle at a time)

walterbux, 2021-11-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

First of all I want to underline that I have tested successfully the bugfix-2.0.x version in which the problem does not exist.

Flashing instead the 2.0.9.2 version with the same extruder configuration I find the following problem:

- if in Repetier Host I select the nozzle number 4 and warm it at operating temperature (say 210 C) also the nozzle number 1 is affected and powered at exactly the same temperatures of the nozzle 4.
- if I repeat the same selecting and powering nozzle 3 then also the nozzle 1 is powered and the temperatures of the 2 nozzles are the same
- a similar thing happens selecting and powering nozzle 2 then also the nozzle 1 is powered at exactly the same temperature of the nozzle 2.

Instead if I select the nozzle 1 and power on each nozzle (1,2,3,4) at nominal temerature and after off  each nozzle behaves normally independently form the others.

Hereafter I attach the 2 configuration files (bugfix and 2.0.9.2 version respectively) but I summarize briefly the kraken extruder configuration:

1 water cooled cube with attached 4 extruders independently managed. each nozzle is apaced 20 mm apart from nozzle 0 (they are positioned at the corners of a square having 20 mm side.
By the way I have not understood how to configure the nozzle positioning in firmware therefore I have done it in the slicers. I ask suggestions for this please.

#define EXTRUDERS 4

#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75

every other thing of extruder section commented out.

Thank you for your work and best regards.

Walter

### Bug Timeline

Found today

### Expected behavior

To manage each extruder nozzle independently from the others.

### Actual behavior

See description.

### Steps to Reproduce

Use Repetier Host control with the 2 firmware versions and attached config files.

### Version of Marlin Firmware

2.0.9.2 version

### Printer model

Custom built cartesian printer similar to Prusa MK3

### Electronics

Mega 2560 Ramps 1.4

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[Marlin-2092-Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7626316/Marlin-2092-Configuration.zip)

thisiskeithb, 2021-11-30:
> First of all I want to underline that I have tested successfully the bugfix-2.0.x version in which the problem does not exist.

Great!

Since the bug is already fixed,it will be resolved in the next stable release (`2.0.9.3`).

closed 2021-11-30 by thisiskeithb

walterbux, 2021-11-30:
Hereafter the Bugfix 2.0.x confiiguration files.
Warmest regards.

Walter
[Marlin-2092-Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7626358/Marlin-2092-Configuration.zip)

walterbux, 2021-11-30:
Please I have also asked help for configuring the kraken extruder nozzle positioning in firmware: how to describe it in FW because there is not any reference to the extruder number... despite of the command 
M218 [T<index>] [X<offset>] [Y<offset>] [Z<offset>] that has it ...
regards

Walter