MarlinFirmware/Marlin #23239: [BUG] Problems with large serial buffers and out-of-order execution

qwertymodo, 2021-12-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Awhile back, I increased the size of the serial input ASCII buffer (MAX_CMD_SIZE 96 and BUFSIZE 32 in Configuration_adv.h) to improve print quality over USB, and after doing so, I noticed a few problems with the increased buffers, due to the way some commands are allowed to skip the buffer and execute out-of-order, while others are not.  The main ones I've noticed this with are temperatures changes and LCD updates, which are allowed to execute out-of-order, and pause/abort commands, which are not.

Allowing temperature commands to execute early can cause a lot of issues, especially with cooldowns at the end of the print, causing underextrusion and even jamming, because the hot end starts to cool off before the print is even complete.  This can be worked around by adding an M400 into your end gcode before the cooldown commands, but that shouldn't need to happen.  I can't think of any reason why temperature changes should be allowed to execute out of order.

On the other hand, pause and abort commands are NOT executed out of order, so with these increased buffer sizes, it can take 30 seconds, or up to almost a minute before the print actually aborts, whether it's requested over USB serial or from the LCD menu.  That's a lot of time for whatever when wrong to keep causing a lot of chaos, or even damage to your hardware.  If you have the ability already built into the serial parser to allow some commands to jump the queue and execute immediately, pause and abort should have been at the absolute top of that list.

Also, LCD screen updates seem to jump the queue as well.  This is more of an annoyance than anything, but I believe these should always happen in order as well.  There are probably other gcode commands where the decision to jump the queue or not should be reexamined, but these are the ones I've noticed.

Also, this may only be the case when printing via USB.  Pausing and aborting, at least, seem to be nearly instantaneous when printing from an SD card.

### Bug Timeline

Old issue, as long as I've used Marlin

### Expected behavior

* I expect a temperature change to occur when it is specified to happen in the gcode file.
* I expect LCD update commands to occur when they are specified to happen in the gcode file.
* I expect a pause or abort command to execute immediately (or as soon as reasonably possible).

### Actual behavior

* Temperature changes occur much earlier than they are supposed to.  This results in the heaters turning off before the print is complete, leading to underextrusion and jams
* LCD updates occur much earlier than they are supposed to, so things like status displays aren't actually in sync with what the printer is actually doing.
* Pause and abort commands take a very long time to actually execute, leaving a long time for things to continue going wrong.  More than once, I've had to resort to hitting the physical power switch to avoid damage to my printer.

### Steps to Reproduce

1. Recompile Marlin with MAX_CMD_SIZE 96 and BUFSIZE 32 in Configuration_adv.h
2. Start a new print sliced with any slicer (don't include an M400 in the end gcode) over USB (I used Octoprint/Octopi)
3. Watch LCD updates like the fan speed or the Z position, they will update well before the printer actually changes to match the displayed values
4. Watch the reported temperature at the end of the print.  The heaters will turn off before the print is completed
5. Attempt to pause or abort the print at any time and watch how long it takes before it actually stops printing.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Creality Ender 3 Pro

### Electronics

BTT SKR Mini E3 v2.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

jaeryang-sindoh, 2021-12-03:
I think you need to change M104(M109).
If you add planner.synchronize() before thermalManager.setTargetHotend() (in M104() or M109() ), you can see what you want. :)

qwertymodo, 2021-12-03:
I just realized that M106 also needs to be synchronized in order for bridge cooling to work.  Otherwise the fan speed changes will happen somewhere other than the bridge.  Looks like right now fans are only synchronized if a certain flag is set related to laser cutters.

qwertymodo, 2021-12-03:
Ok, so it's probably worth going through the full list of gcode commands and listing the ones that need to be synced and which ones don't.  Here's the ones that I see (or at least the ones that should be set a certain way, I haven't checked the current behavior on any of these yet, but I'll go through them when I get a chance).  If anybody can think of any others, I'll add them to the list here.  Also, I'm coming at this from a 3D printer perspective, these may be different for other devices like laser printers.

Commands that *should* synchronize before executing:
- M73: Set Print Progress
- M104: Set Hotend Temperature
- M106: Set Fan Speed
- M107: Fan Off
- M109: Wait for Hotend Temperature
- M117: Set LCD Message (maybe... I can see an argument for synchronizing or not)
- M140: Set Bed Temperature
- M141: Set Chamber Temperature
- M190: Wait for Bed Temperature
- M191: Wait for Chamber Temperature
- M192: Wait for Probe Temperature
- M220: Set Feedrate Percentage
- M221: Set Flow Percentage

Commands that *should not* synchronize before executing:
- G27: Park Toolhead
- M0/M1: Unconditional Stop
- M25: Pause SD Print
- M76: Pause Print Job
- M105: Report Temperatures
- M112: Emergency Stop
- M125: Park Head
- M290: Babystep
- M410: Quickstop
- M486: Cancel Objects
- M524: Abort SD Print
- M600: Filament Change

qwertymodo, 2021-12-05:
I've updated the list in my previous comment to remove gcode commands that currently behave as expected.  Unfortunately, after doing so, I can't find where the delay in aborting prints over USB is coming from, maybe somebody else has an idea there.

jaeryang-sindoh, 2021-12-07:
When aborting, M410 will stop printing immediately. Except delay when cleaning buffer(cleaning_buffer_counter = TEMP_TIMER_FREQUENCY) and restart the block delay(delay_before_delivering = BLOCK_DELAY_FOR_1ST_MOVE), I think there is no delay when aborting.

qwertymodo, 2021-12-07:
I'll have to do some more digging to see exactly what command(s) Octoprint sends when pausing/aborting from the web interface, but it definitely has a significant delay.

ellensp, 2021-12-07:
In you increase buffer to an absurd amount you will have to wait an absurd amount of time for any inserted gcode to be processed, Unless it it handled by the emergency passer and the emergency passer is enabled

qwertymodo, 2021-12-07:
I do have the emergency parser enabled.  It still takes ~30s or so to abort the print after clicking abort in OctoPrint.

ellensp, 2021-12-07:
try just M112 from the console

qwertymodo, 2021-12-09:
Ok, so the abort delay might actually be something to be solved host-side.  I think Octoprint doesn't actually issue any "abort" command, it just stops sending new gcode, so the printer continues to complete all of the queued commands currently in the buffer.  Is there a gcode command that can be read from the emergency parser to empty out the planner queue?  That would allow for more advanced post-abort handling, such as lifting/parking the head and initiating cooldown, without requiring a full reset of the machine like with M112.

As far as other gcode commands go, after watching some prints more carefully, it seems like fan speed changes are properly synchronous, so bridge cooling works as expected.  That probably  just leaves temperature changes and LCD updates that seem to improperly jump the queue.

jaeryang-sindoh, 2021-12-09:
Yes. Octoprint's abort is that gcode sending is just stopped. 

When Marlin receives M410, Marlin will stop motor immediately(interrupt disable) and empty buffer in TEMP_TIMER_FREQUENCY.

M112 will reset Marlin.

qwertymodo, 2021-12-09:
Ok, adding an M410 to the beginning of OctoPrint's post-abort script seems to have achieved the desired result of aborting immediately, cancelling any buffered commands, and then continuing with the rest of the post-abort script (lift head, re-home X/Y, park head, shut off heaters, present build plate, etc).  So that's one more in the "not a bug" pile.  Just need to figure out temperature changes and LCD updates.

qwertymodo, 2021-12-10:
I just tried pausing a print via the LCD menu, and there was a delay there as well, so that one isn't due to OctoPrint.

qwertymodo, 2021-12-10:
I went ahead and tried recompiling with @jaeryang-sindoh 's suggested addition of planner.synchronize(), and although it does seem to achieve the intended result, in that the temperature changes now occur when they're supposed to instead of early, it also seems to be causing a bunch of short "glitches" where the temperature is getting changed and then back again very rapidly.  Here's a graph as I'm printing a calibration temperature tower.

[image]

ellensp, 2021-12-10:
You disable all buffering, you get movement glitches as the buffer is refilled....

qwertymodo, 2021-12-10:
Those aren't movement glitches, they're temperature glitches.

jaeryang-sindoh, 2021-12-10:
It is strange. I have never seen this. Maybe it is Octoprint's bug? :)

qwertymodo, 2021-12-10:
Tomorrow when I have a chance, I'll try printing this same gcode from an SD card and see if the same temperature glitches occur.

closed 2022-03-27 by github-actions[bot]