MarlinFirmware/Marlin #23241: Compiled file fails at the last second

uggels, 2021-12-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Im new to 3d printing and even newer with editing software. I have an Ender 5 Plus with an BTT Octopus V1_1 and a BTT tft35 v3.0. ive been having trouble being successful in compiling my first firmware. I did everything i could to get rid of the error messages but there still seems to be an issue and im not entirely sure how to fix it. Any kind of advice would be helpfu
[image]
l.

### Bug Timeline

im not sure if this has been an issue in the past

### Expected behavior

i expected my firmware to be compiled seeing as i looked up as much information as i could to get it to run. 

### Actual behavior

at the last moment before the compilation is over, it fails and the error message is nothing i can find info on.

### Steps to Reproduce

1. dowloaded marlin, BTT octopus firmware, and configuration file
2. inputted all the required setting required to get the printer to work based on tutorials i found for my specific equipment
3. compiled
4. failed

### Version of Marlin Firmware

Marlin-bugfix-2.0x 

### Printer model

ender 5 plus

### Electronics

Bigtreetech Octopus V1.1

### Add-ons

BLTouch, BBT TFT35 V3.0, 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7631673/Configuration.zip
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/7631674/Configuration_adv.zip)
)

ellensp, 2021-12-01:
When did you download bugfix? as these error where meant to have been fixed 11 hours ago

tpruvot, 2021-12-01:
its different, press the Clean button before building...

ellensp, 2021-12-01:
Clean build on current bugfix using provided Config files compiles (and links, as this error is a linking error) without issues.

closed 2021-12-01 by ellensp

uggels, 2021-12-01:
> 

Haha I guess I downloaded the files a day too early. I just downloaded the newer version and I worked great thank you so much for the help