MarlinFirmware/Marlin #23242: babystepping working automatically

muhammetbaygin, 2021-12-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

while printing I go into the ministep tab and raise the nozzle by 1mm. and when I exit the screen it automatically rises 1mm again.

### Bug Timeline

Always. update marlin 2.0.9.2

### Version of Marlin Firmware

2.0.9.2

### Printer model

özel

### Electronics

yükseltme kartı

Sonicboom247, 2021-12-02:
do you have //#define MOVE_Z_WHEN_IDLE commented out?

muhammetbaygin, 2021-12-02:
yes

muhammetbaygin, 2021-12-30:
is there anyone who can help?

ellensp, 2021-12-30:
nothing here to go on. No Configuration files. 
No mention of Motherboard or display types. So no way to try and reproduce.

muhammetbaygin, 2021-12-30:
motherboard: mks sgen L
display: tft24 gd bigtreetech
printer same ender 3

closed 2022-03-24 by github-actions[bot]