MarlinFirmware/Marlin #23245: [BUG] Polargraph broken

dutchco, 2021-12-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Using the latest firmware ( bugfix 2.0.x ), and the accordingly generated polargraph example file.

Whenever i start a 'print', my machine ( makeangelo DIY ) homes perfectly fine.
However, when the first movement is sent to the machine, it just moves the the center, and stops moving all together after that.
From that point on, it just moves the servo up and down for a few minutes, and then i start seeing errors in my console saying 'Home XY first'. When this happens, my controller ( Ramps 1.4 ) crashes completely and reboots.

### Bug Timeline

It started after i upgraded to the latest marlin bugfix 2.0.x firmware. Makelangelo's firmware revision works fine, but also crashes sometimes.

### Steps to Reproduce

1. run this piece of Gcode on a polargraph running the latest marlin bugfix firmware.
`; 2021-11-30 at 14:23:33 CET
G28 X Y 
M280 P0 S30 T50
G1 X-85.844 Y84.281 F7500.0
G1 X-86.119 Y83.995 F7500.0
G1 X-86.381 Y83.736 F7500.0
G1 X-86.631 Y83.505 F7500.0
G1 X-86.869 Y83.301 F7500.0
G1 X-87.094 Y83.125 F7500.0
G1 X-87.094 Y83.125 F7500.0
G1 X-87.314 Y82.971 F7500.0
G1 X-87.539 Y82.835 F7500.0
G1 X-87.767 Y82.716 F7500.0`

### Version of Marlin Firmware

Bugfix 2.0.x ( cloned today )

### Printer model

Polargraph

### Electronics

Ramps 1.4

### Host Software

Pronterface

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7635636/Marlin.zip)

thinkyhead, 2021-12-16:
CC: @i-make-robots Ref: #22790

i-make-robots, 2021-12-23:
It sounds like something in your config is wrong, which causes all inverse kinematic calculations to return (0,0,z), which makes all moves drive to center.  Without seeing your config I couldn't say more.  This is (the fork I use)[https://github.com/MarginallyClever/Marlin-polargraph/tree/polargraph] and (a helpful guide is here)[https://www.marginallyclever.com/2021/10/friday-facts-4-how-to-marlin-polargraph/]

dutchco, 2021-12-23:
Hi,

Thanks for the reply! I appreciate it a lot. I really like the idea of having a marlin powered polargraph, since I can remotely control it using octoprint on the new raspberry pi zero!

I resolved the issue with the crashing arduino: the servo caused a power drop. So that's fixed and not a firmware issue.

However, i took the config that was provided in the examples, and I also followed the steps precisely and made a new config, but it still bugged out for some reason. It just goes to one position, and stays there just making servo movements. The console says I need to home X/Y first. 

The same config changes work perfectly for the branch you provide by the way!

I'll add you on discord tomorrow, since we're in a different timezone and I'm going to get some sleep.

i-make-robots, 2021-12-23:
Well if it says home x/y.... home it.

closed 2022-03-27 by github-actions[bot]