MarlinFirmware/Marlin #23248: [BUG] Stepper motors are noisy/vibrate at low speeds when StealthChop is active

magnificu, 2021-12-02:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Stepper motors are noisy/vibrate at low, medium and high speeds (30mm/s, 60mm/s, 80mm/s, 120mm/s) when StealthChop is active
The stepper divers are used in SPI mode.
If StealthChop is deactivated the motors are not vibrating anymore

**Configuration_adv.h**
#define TMC_USE_SW_SPI
#define STEALTHCHOP_XY
#define CHOPPER_TIMING CHOPPER_DEFAULT_24V 

Updating the CHOPPER_TIMING it smoothing the noise when StealthChop is deactivated.
With the StealthChop active the motors are noisy/vibrating again
#define CHOPPER_TIMING_X  {3, 6, 8}
#define CHOPPER_TIMING_Y  {3, 6, 8} 

### Expected behavior

When StealthChop is active the stepper motors should be quieter

### Actual behavior

When StealthChop is active the stepper motors are very noisy/vibrating

### Steps to Reproduce

Upload firmware Marlin 2.0.9.2 and print

### Version of Marlin Firmware

Marlin 2.0.9.2

### Printer model

coreXY custom

### Electronics

Power supply 24V 16.5A, SKR2 board, BTT TMC5160 v1.2, X & Y Stepper motors: 1.8° QSH4218-47-28-040

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

I tested as well with another 12V power supply
I tested as well with stepper motors 17HM19-2004 (0.9° stepper motor with updated config to CHOPPER_09STEP_24V)
The results were worse.

The previous version I used was Marin 1.1.9
Ramps_v1.6
TMC2208
unknown 1.8° stepper motors
and the motors were very smooth/silent

XDA-Bam, 2021-12-02:
Please attach your current configs, otherwise diagnostics will be difficult.

magnificu, 2021-12-02:
> Please attach your current configs, otherwise diagnostics will be difficult.
[Marlin_configs.zip](https://github.com/MarlinFirmware/Marlin/files/7643625/Marlin_configs.zip)

magnificu, 2021-12-02:
@XDA-Bam 
> > [Marlin_configs.zip](https://github.com/MarlinFirmware/Marlin/files/7643625/Marlin_configs.zip)

XDA-Bam, 2021-12-02:
Did you test `#define SQUARE_WAVE_STEPPING`? May be worth a try.

magnificu, 2021-12-02:
I was skeptical because it's a beta feature, but I'll give it a try and let you know if something improves.

magnificu, 2021-12-02:
I test it and I did not noticed any difference in SpreadCycle (StealthChop disabled)
In StealthChop same noise/vibration of the motors is present.

GhostlyCrowd, 2021-12-02:
> I test it and I did not noticed any difference in SpreadCycle (StealthChop disabled) In StealthChop same noise/vibration of the motors is present.

and this is on latest bugfix as well? have you tested that?

magnificu, 2021-12-02:
@GhostlyCrowd
I just checked the latest bugfix and the issue is the same.

magnificu, 2021-12-05:
Additional data that might help:

Yellow - Current
Purple - Voltage measured on the Stepper coil

[image]
[image]
[image]

magnificu, 2021-12-05:
Additional data that might help on the debug:
Yellow -> Current measured with a current clamp
Purple -> Voltage measured on one coil of the stepper

Spread Cycle
[image]

StealthChop
[image]

[image]

npmomchev, 2021-12-15:
I confirm the problem is present in the latest error correction. Board: BTT-SKR-pro with TMC5160 drivers Steppers: NEMA17 Moons. I tried NEMA17 with 0.9 degree step and NEMA17 with 1.8 degree step

magnificu, 2021-12-17:
At the moment I am using Marlin with StealthChop disabled, but of course this is a limitation.

Marcos2BR, 2021-12-27:
I can't take tests right now , I'm going away , but just to report: 
hypercube evolution : skr 1.4 + tmc2209 + 17HS8401
installed 2.0.9.3 using same tweaks as 2.0.9.2 ,
and my motors started vibrating too , i printed diagonal lines at 50mms and they ghosted while the motor vibrated .
before, everything worked fine.

magnificu, 2022-01-11:
Is anyone looking into this?

magnificu, 2023-06-28:
TMC5160 are becoming more and more popular and I hope with feedback from more users we can find a solution.
In my opinion, this is a software issue.

thisiskeithb, 2023-06-28:
Can you upload a video demonstrating the noise? I run BTT's 5160s on a couple boards here (STM32F4 based like the SKR 2 & BTT002) and I wouldn't consider the noise to be out of the norm.

magnificu, 2023-06-28:
It can also be seen from the oscilloscope that the sine wave is highly distorted in StealthChop compared to Spread Cycle.
I will do a video as well.

magnificu, 2023-07-01:
Is anybody that can really help me with this topic?

magnificu, 2023-07-02:
Could it be that the problem is that QSH4218-47-28-040 stepper motors have an ultra low phase inductance of 0.6mH?

magnificu, 2023-07-03:
I've changed the TCM5160 (SPI) drivers with TMC2208 (UART) and the noise is gone.
I think it is something wrong on how TMC5160 is coded in Marlin.

Rincewind08, 2023-07-03:
I haven't been able to use the BTT 5160s on my GTR Boards in several
months, nor my BTT Octopus v1.1, nor my MKS Monster8 V2.  With or without
my clock pins clipped. Also, a cheapo generic Step stick 5160 fails to boot
as well. What change did you make to your configurations to make your board
boot?  Did you encounter any similar issues?  I've also tried relentlessly
with my SKR v1.3s, v1.4 as well as a 1.4T.  I don't have any success in
activating a board with configurations set for these drivers, despite
having been able to run them initially?

magnificu, 2023-07-03:
In my case they are working, but noisy in StealthChop mode.
You can check the configuration I attached and compare them to yours using software like Meld.
BTT TMC5160 v1.2 with SKR2 does not require CK pin to be clipped. Read the documentation and check if you set the jumpers correctly.
Anyway the topic here is different.

fjulian79, 2024-09-03:
Since I have reported the issue once again in #27403 and triggered reopening this one, I investigated further to figure out what happens when those drivers get initialized. I have learned that the TMC drivers in general perform an automatic tuning or lets call it adoption to the used motor and also the mechanical load driven by the motor. They even adopt to dynamic load conditions and speeds. What if the issue is caused by a faulty TMC tuning in some cases??

I would assume that magnificu and myself are not the only ones who bought TMC5160 drivers but the web isn't full of users reporting this issue so I must expect that is not a general issue. On the other side, there is black magic auto tuning which is definitely influenced by the particular machine. Maybe, if some factors add up, it might end up in the situation reported here. 

There are also two old shorts on YouTube, see ..
https://www.youtube.com/shorts/TXC0VF2c5tE
https://www.youtube.com/shorts/jFp_4yF6a1Q
The guy who posted them writes about exchanging the sense resistor but nothing more to work with. Anyway .. this is a little bit more extreme example of what happens here. I also observe that the is issue is more dominant at the X axis.

When I lay my finger on the printhead during noisy movements I can feel each step of the motor. I take this as indicator for a to high current setting. But changing the current setting does not solve the issue .. 800mA was my start value, went down to 500mAh, there it was better but still not solved and the generated forces where just high enough to print with 60mm/s so this is not usable.

I have also checked the TMC Registers .. 
```
After Power Up
Send: M122 V
Recv:         X        Y        Z        E
Recv: GCONF         0x00:00:00:04    0x00:00:00:04    0x00:00:00:04    0x00:00:00:04
Recv: IHOLD_IRUN    0x00:0A:0F:07    0x00:0A:0F:07    0x00:0A:0F:07    0x00:0A:0F:07
Recv: GSTAT         0x00:00:00:05    0x00:00:00:05    0x00:00:00:05    0x00:00:00:05
Recv: IOIN          0x30:00:00:52    0x30:00:00:52    0x30:00:00:52    0x30:00:00:50
Recv: TPOWERDOWN    0x00:00:00:80    0x00:00:00:80    0x00:00:00:80    0x00:00:00:80
Recv: TSTEP         0x00:0F:FF:FF    0x00:0F:FF:FF    0x00:0F:FF:FF    0x00:0F:FF:FF
Recv: TPWMTHRS      0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: TCOOLTHRS     0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: THIGH         0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: CHOPCONF      0x14:00:82:84    0x14:00:82:84    0x14:00:82:84    0x14:00:82:84
Recv: COOLCONF      0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: PWMCONF       0xC8:0D:0E:24    0xC8:0D:0E:24    0xC8:0D:0E:24    0xC8:0D:0E:24
Recv: PWM_SCALE     0x00:00:00:07    0x00:00:00:07    0x00:00:00:07    0x00:00:00:07
Recv: DRV_STATUS    0x80:07:40:00    0x80:07:40:00    0x80:07:40:00    0x80:07:40:00

After first auto home
Send: M122 V
Recv:         X        Y        Z        E
Recv: GCONF         0x00:00:00:04    0x00:00:00:04    0x00:00:00:04    0x00:00:00:04
Recv: IHOLD_IRUN    0x00:0A:0F:07    0x00:0A:0F:07    0x00:0A:0F:07    0x00:0A:0F:07
Recv: GSTAT         0x00:00:00:05    0x00:00:00:05    0x00:00:00:05    0x00:00:00:05
Recv: IOIN          0x30:00:00:42    0x30:00:00:40    0x30:00:00:42    0x30:00:00:50
Recv: TPOWERDOWN    0x00:00:00:80    0x00:00:00:80    0x00:00:00:80    0x00:00:00:80
Recv: TSTEP         0x00:0F:FF:FF    0x00:0F:FF:FF    0x00:0F:FF:FF    0x00:0F:FF:FF
Recv: TPWMTHRS      0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: TCOOLTHRS     0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: THIGH         0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: CHOPCONF      0x14:00:82:84    0x14:00:82:84    0x14:00:82:84    0x14:00:82:84
Recv: COOLCONF      0x00:00:00:00    0x00:00:00:00    0x00:00:00:00    0x00:00:00:00
Recv: PWMCONF       0xC8:0D:0E:24    0xC8:0D:0E:24    0xC8:0D:0E:24    0xC8:0D:0E:24
Recv: PWM_SCALE     0x00:01:00:0E    0x00:00:00:0E    0x00:00:00:0C    0x00:00:00:07
Recv: DRV_STATUS    0x80:07:40:00    0x80:07:40:00    0x80:07:40:00    0x80:07:40:00
```
You can see that something in PWM_SCALE has changed

Back to the auto tune magic: In the marlin code I have seen that it would be possible to disable the black unicorn TMC magic. See src/module/stepper/trinamic.cpp starting in line 833
```
TMC2160_n::PWMCONF_t pwmconf{0};
pwmconf.pwm_lim = 12;
pwmconf.pwm_reg = 8;
pwmconf.pwm_autograd = true;
pwmconf.pwm_autoscale = true;
pwmconf.pwm_freq = 0b01;
pwmconf.pwm_grad = 14;
pwmconf.pwm_ofs = 36;
st.PWMCONF(pwmconf.sr);
```
At the end I was not sure about what will happen if I force autograd and autoscale to false .. so I have not just tried it.

Maybe this input helps to clarify what is going on here .. maybe not. 
My intention was not to take my input as final conclusion and close the issue again ;-)

I have ordered right now a set of TMC2209 drivers .. at first for a test and if this one here is not solved within a reasonable amount of time (which can be expected because of the non straight forward character) I will use them instead of my 5160's. However .. would be a pity to leave the 5160's behind as they are not usable.

Meina88, 2024-12-04:
Hi! My recommendation is to change the sense resistors. I have made my own drivers based on TMC5160 and with 0.75R sense resistors you will get vibrations on small motors. They will work great on motors requiring 3 or 4A RMS. But if you want to run smaller motors, use 0.10, 0.12 or 0.15 sense resistors. Do your own testing. Use the table in chapter 9 of the datasheet.

magnificu, 2024-12-05:
The problem is that the TMC5160 is built to handle high currents, around 4A when using the stock Rsense resistors with a value of 0.075 Ohms. When the drivers are set (in Marlin) to provide lower currents, around 2A, the motors produce a lot of noise. I modified my drivers by replacing both stock Rsense resistors (0.075 Ohms) with 0.15 Ohms, and the noise disappeared.

closed 2025-03-20 by github-actions[bot]