MarlinFirmware/Marlin #23249: [FR] implement CoolStep – Load Adaptive Current Control

Sonicboom247, 2021-12-02:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

No

### Describe the feature you want

As described in the TMC2209 datasheet section 1.6 adding CoolStep – Load Adaptive Current Control would be a massive step forward in noise and power reduction in our systems.

coolStep drives the motor at the optimum current. It uses the stallGuard4 load measurement
information to adjust the motor current to the minimum amount required in the actual load situation.
This saves energy and keeps the components cool.
Benefits are:
- **Energy efficiency power consumption decreased up to 75%**
- Motor generates less heat improved mechanical precision
- **Less or no cooling improved reliability**
- Use of smaller motor less torque reserve required → cheaper motor does the job
- Less motor noise Due to less energy exciting motor resonances

https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf

thisiskeithb, 2021-12-02:
There was an unsuccessful effort to implement this and in the end, it was reverted.

See [these PRs](https://github.com/MarlinFirmware/Marlin/pulls?q=is%3Apr+coolstep+is%3Aclosed+in%3Atitle) for more information.

closed 2021-12-02 by thisiskeithb

Sonicboom247, 2021-12-02:
Thanks, hopefully it will get resurrected in the future