MarlinFirmware/Marlin #23252: [FR] GCODE to send CHOPPER_TIMING {off_time, hysteresis_end, hysteresis_start}

magnificu, 2021-12-02:
### Is your feature request related to a problem? Please describe.

I have noise/vibration with the stepper motors and it would be great to send the CHOPPER_TIMING via GCODE so that I can check on the oscilloscope the distortions of the sin wave.

### Describe the feature you want

In order to optimize quickly the CHOPPER_TIMING for different stepper drivers and motors it would be great to be able to send the CHOPPER_TIMING {blank time TBL, off_time, hysteresis_end, hysteresis_start} via GCODE

Is there any easy way to implement it? Where should I look/start?

magnificu, 2021-12-03:
I added to "gcode.cpp" (row 1044 in Marlin 2.0.9.2)
      #if ENABLED(CHOPPER_TIMING_GCODE)
        case 9999: M9999(); break;                                  // M9999: set CHOPPER_TIMING via GCODE
      #endif   

I added to "gcode.h" (row 1185 in Marlin 2.0.9.2)
  #if ENABLED(CHOPPER_TIMING_GCODE)
    static void M9999();
  #endif

I created a new file Marlin/src/gcode/M9999.cpp with this content

#include "../gcode.h"
#include "../../module/motion.h"

#include "../../MarlinCore.h"

/**
 * M9999: O3, E6, S8 (off_time[1..15]>, <hysteresis_end[-3..12], hysteresis_start[1..8])
 */
void GcodeSuite::M9999() {
  int ot, he, hs;
  ot = parser.seenval('O');
  he = parser.seenval('E');
  hs = parser.seenval('S');
}

How can I pass the ot, he and hs variables to "trinamix.cpp" ?
    chopconf.toff = ot;
    chopconf.hend = he + 3;
    chopconf.hstrt = hs - 1;

oliver-eifler, 2021-12-03:
you have to use tmc driver class functions to set/changes these values on the fly
blank_time(config.blank_time);
toff(config.off_time);
etc.
have a look at tmcstepper lib examples 
https://github.com/teemuatlut/TMCStepper/tree/master/examples/Calibrate_spreadCycle

and class TMCMarlin in marlin source
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/tmc_util.h

magnificu, 2021-12-03:
Thank you for your feedback.
I will give it a try, even if this seems to be more complex than my C programing skills.

magnificu, 2021-12-05:
Any help on how to set/change on the fly the chopper_time would be much appreciated.

magnificu, 2021-12-05:
I found in trinamic.cpp this function, but I have issues in understanding and passing all the variables:

void tmc_init(TMCMarlin<TMC5160Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID> &st, const uint16_t mA, const uint16_t microsteps, const uint32_t hyb_thrs, const bool stealth, const chopper_timing_t &chop_init, const bool interpolate) {
    st.begin();

How to ensure that I use the same variables set in the configuration.h & configuration_adv.h files?

From where I get the ID's?
AXIS_LETTER, DRIVER_ID, AXIS_ID> &st

I need an example and some help with which headers to includ

oliver-eifler, 2021-12-06:
analyze void GcodeSuite::M906() function in Marlin\src\gcode\feature\trinamic\M906.cpp
it calls rms_current(...) setter from TMCMarlin for every Axis according to parameters of m-code M906

you have to extend the TMCMarlin class with functions to get/set the chopper values

As this part marlin uses C++ classes and C++ Templates you'll need some knowledge how C++ works - no complicated things like STL (standard template library) but the basics

magnificu, 2021-12-06:
Sorry if my question does not make sense to you.
How the #define CHOPPER_TIMING CHOPPER_DEFAULT_24V from Configuration_adv.h are send to the TMC drivers?
I thought this functionality is already implemented and I just have to "call" it, but with values given in the GCODE rather than in the Configuration_adv.h file.
CHOPPER_DEFAULT_24V is actually defined in "tmc_util.h" #define CHOPPER_DEFAULT_24V  { 4,  2, 1 }

...to extend the TMC class is beyond my capabilities.
I will let it to somebody who knows if considers that this feature worth to be implemented.
I think in combination with an octoprint python script or BTT TFT tuning can be a nice functionality to easily tune the chopper time

Thank you for your feeback

magnificu, 2021-12-09:
I managed to read the chopper times using a gcode command:

I added to "tmc_util.h"
-------------------------------------------------------------------------------------------
(row 312 in Marlin 2.0.9.2)
template<typename TMC>
void tmc_print_chopper_time(TMC &st) {
  st.printLabel();
  SERIAL_EOL();
  SERIAL_ECHOLNPGM(" chopper blank time: ", st.blank_time());
  SERIAL_ECHOLNPGM(" chopper toff: ", st.toff());
  SERIAL_ECHOLNPGM(" chopper hysteresis_end: ", st.hysteresis_end());
  SERIAL_ECHOLNPGM(" chopper hysteresis_start: ", st.hysteresis_start());
}

I created a new file Marlin/src/gcode/M9999.cpp with the content:
-------------------------------------------------------------------------------------------
void GcodeSuite::M9999() {
  #define TMC_SAY_CHOPPER_TIME(Q) tmc_print_chopper_time(stepper##Q)
  
  bool report = true;

  #if AXIS_IS_TMC(X) || AXIS_IS_TMC(X2) || AXIS_IS_TMC(Y) || AXIS_IS_TMC(Y2) || AXIS_IS_TMC(Z) || AXIS_IS_TMC(Z2) || AXIS_IS_TMC(Z3) || AXIS_IS_TMC(Z4) || AXIS_IS_TMC(I) || AXIS_IS_TMC(J) || AXIS_IS_TMC(K)
    const uint8_t index = parser.byteval('I');
  #endif

  if (report) {
    #if AXIS_IS_TMC(X)
      TMC_SAY_CHOPPER_TIME(X);
    #endif
    #if AXIS_IS_TMC(X2)
      TMC_SAY_CHOPPER_TIME(X2);
    #endif
    #if AXIS_IS_TMC(Y)
      TMC_SAY_CHOPPER_TIME(Y);
    #endif
    #if AXIS_IS_TMC(Y2)
      TMC_SAY_CHOPPER_TIME(Y2);
    #endif
    #if AXIS_IS_TMC(Z)
      TMC_SAY_CHOPPER_TIME(Z);
    #endif
    #if AXIS_IS_TMC(Z2)
      TMC_SAY_CHOPPER_TIME(Z2);
    #endif
    #if AXIS_IS_TMC(Z3)
      TMC_SAY_CHOPPER_TIME(Z3);
    #endif
    #if AXIS_IS_TMC(Z4)
      TMC_SAY_CHOPPER_TIME(Z4);
    #endif
    #if AXIS_IS_TMC(I)
      TMC_SAY_CHOPPER_TIME(I);
    #endif
    #if AXIS_IS_TMC(J)
      TMC_SAY_CHOPPER_TIME(J);
    #endif
    #if AXIS_IS_TMC(K)
      TMC_SAY_CHOPPER_TIME(K);
    #endif
    #if AXIS_IS_TMC(E0)
      TMC_SAY_CHOPPER_TIME(E0);
    #endif
    #if AXIS_IS_TMC(E1)
      TMC_SAY_CHOPPER_TIME(E1);
    #endif
    #if AXIS_IS_TMC(E2)
      TMC_SAY_CHOPPER_TIME(E2);
    #endif
    #if AXIS_IS_TMC(E3)
      TMC_SAY_CHOPPER_TIME(E3);
    #endif
    #if AXIS_IS_TMC(E4)
      TMC_SAY_CHOPPER_TIME(E4);
    #endif
    #if AXIS_IS_TMC(E5)
      TMC_SAY_CHOPPER_TIME(E5);
    #endif
    #if AXIS_IS_TMC(E6)
      TMC_SAY_CHOPPER_TIME(E6);
    #endif
    #if AXIS_IS_TMC(E7)
      TMC_SAY_CHOPPER_TIME(E7);
    #endif
  }
}
[image]
[image]
[image]

magnificu, 2021-12-09:
I am stuck in understanding how to write the chopper times and I face issues with the data types
e.g.
    if (parser.seenval('O')) {
      int16_t toff;
      toff = parser.value_byte(); //--> error here uint8_t data type required by TMC5160 that reuses class TMC2130Stepper from TMCStepper.h
      SERIAL_ECHOPGM_P(" toff=", toff);
      SERIAL_EOL();
    }

Can somebody create a simple example on how to overwrite the chopper time?
I think is not difficult if you know how Marlin works...

Anyway I wanted to share with you the progress I did after a lot of struggle with the classes reuse mess...

magnificu, 2021-12-10:
I have a first working draft of the code and I am doing some cleanups.
Can someone instruct me how can I prepare and send a commit with the code?
Which gcode number I can use for this? At the moment I am using M9999.
Is there any rule on setting the gcode parameters?

The new gcode works like this e.g.:
M9999
read all chopper times for all axis

M9999 X B24 O3 N6 S8
sets for the X axis 
blank_time=24
time_off=3
hysteresis_end=6
hysteresis_start=8

e.g. to set Z2 chopper times
M9999 I1 Z B24 O3 N6 S8

e.g. to set E0 chopper times
M9999 I0 E B24 O3 N6 S8

For tuning the chopper times now a gcode can be created to move the axis at different speeds and test with different chopper times.

XDA-Bam, 2021-12-18:
> I have a first working draft of the code and I am doing some cleanups. (I test it only for TMC5160) Can someone instruct me how can I prepare and send a commit with the code?

You have to fork MarlinFirmware/Marlin. Then you create a new branch based on the most recent bugfix-2.0.x. In that branch, you apply all you changes / commits and push them. Then, you login to github.com and navigate to your forked Marlin repository - on the top the main repository page, there should be a message in the form of "You pushed XYZ to magnificu/Marlin. Do you want to create a pull request?". Click on the corresponding button to create that pull request. Make sure you select "bugfix-2.0.x" as remote branch in the next step and submit.

closed 2021-12-22 by magnificu

magnificu, 2022-01-05:
https://marlinfw.org/docs/gcode/M919.html