MarlinFirmware/Marlin #23254: Marlin Bugfix 2.0.9.2 fails to build on Debian 11 using VSCode 1.62.3

oliverhbailey, 2021-12-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Attempting to do a test build with the default configuration files produces the following errors. This is a new VSCode install and the Marlin bugfix 2.0.x update as of December 2nd, 2021. The errors follow:

ModuleNotFoundError: No module named 'platformio.managers.package':
  File "/home/oliverhbailey/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 178:
    env.SConscript(item, exports="env")
  File "/home/oliverhbailey/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/oliverhbailey/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/oliverhbailey/Downloads/Marlin-bugfix-2.0.x/buildroot/share/PlatformIO/scripts/common-features-dependencies.py", line 12:
    from platformio.managers.package import PackageManager

### Bug Timeline

Satrted upon installing yesterday on Debian 11 with fresh VSCode install and fresh install of marlin 2.0.x bugfix downloaded December 2nd, 2021

### Expected behavior

Expected results were a successful test build for installation verification. Python 3 is installed on the target machine.

### Actual behavior

Following Errors producted almost immediately upon starting the buld process

ModuleNotFoundError: No module named 'platformio.managers.package':
  File "/home/oliverhbailey/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 178:
    env.SConscript(item, exports="env")
  File "/home/oliverhbailey/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/oliverhbailey/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/oliverhbailey/Downloads/Marlin-bugfix-2.0.x/buildroot/share/PlatformIO/scripts/common-features-dependencies.py", line 12:
    from platformio.managers.package import PackageManager

### Steps to Reproduce

1 Install VSCode
2 Install Marlin Bugfix
3 Run autobuild to test installation

### Version of Marlin Firmware

2.0.x bugfix downloaded 12-2-2021

### Printer model

Ender 3 Max

### Electronics

Stock

### Add-ons

BLTouch

### Bed Leveling

ABL Linear grid

### Host Software

Other (explain below)

### Additional information & file uploads

Cannot get test build with default downloaded files.

ellensp, 2021-12-03:
buildroot/share/PlatformIO/scripts/common-features-dependencies.py does not exist in Marlin
see the directory here https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/buildroot/share/PlatformIO/scripts

ellensp, 2021-12-03:
I can only find one refernce to this, a 3rd party repository https://github.com/jmz52/Marlin/blob/R61502/buildroot/share/PlatformIO/scripts/common-features-dependencies.py
Which is not genuine marlin and not supported here

closed 2021-12-03 by ellensp