MarlinFirmware/Marlin #23255: [BUG] Longer LK5 Pro - Y_Min_endstop not working

Ruiaxe, 2021-12-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm upgrading the stock firmware of the printer to Marlin 2.0.9.2.

Using the provided board BOARD_LONGER3D_LKx_PRO the Y endstop always shows in M119 as open, if when the endstop is pressed.

Also Y movement is strange, it starts moving when homing Y but it stops moving under 1 second, very far away from the endstop.

Reverting to stock firmware all works fine again.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7650786/Marlin.zip)

### Bug Timeline

Its new.

### Expected behavior

I expected the Y axis to move, bump the endstop and home correctly.

### Actual behavior

It doesn't home Y.

X and Z (with bltouch) work fine.

### Steps to Reproduce

1st Step: G28 Y

### Version of Marlin Firmware

2.0.9.2

### Printer model

Longer Lk 5 Pro

### Electronics

Stock

### Add-ons

bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Additional information & file uploads

What I did to try to correct/diagnose the issue:

- Confirmed the Y_MIN_ENDSTOP_INVERTING logic comparing with the old firmware definitions provided by Longer;
- Confirmed the pins in the board pins_LONGER3D_LKx_PRO.h, and looks ok;
- Checked m119 outputs, with no change when the endstop is pressed;
- Checked M43 report to verify if the pins where being taked, and yes they where;
- Uncommented also the USE_XMAX_PLUG, USE_YMAX_PLUG, USE_ZMAX_PLUG to check if there were changes in m119 to account for incorrect pins taken, but all was ok;
- Reverted back to old firmware and confimed I can G28 correctly.

Also a additional information, this is the first time I'm doing this, so I might have overlooked something very basic. I that is correct I'm sorry in advance. I do believe that I explored all the avenues I could taken in account my limited knowledge in the matter. I investigated the matter before posting and it seams to me a bug.

Thanks for your time on this.

ellensp, 2021-12-03:
This boards pin file has a odd switch
If you have any serial port set to port 3 or have manually enabled CHANGE_Y_LIMIT_PINS
The board uses Z-MAX plug as Y-min 

Or the board is already wired like this and you need to enable CHANGE_Y_LIMIT_PINS or more the endstop plug..

Ruiaxe, 2021-12-04:
Please reopen the issue.

I compiled without the Display and with work correctly:

//#define DGUS_LCD_UI_RELOADED

Thanks for your time on this.

Ruiaxe, 2021-12-04:
To be more precise I add to disable more functions in order to compile without display:

Disabled:
//#define DGUS_LCD_UI_RELOADED
//#define FILAMENT_RUNOUT_SENSOR

Adv
//#define ADVANCED_PAUSE_FEATURE

ellensp, 2021-12-05:
real output of M43

> PIN:  14   Port: J1        TXD3                                   protected 
> .                          Y_MIN_PIN                              protected 
> .                          Y_STOP_PIN                             protected 

Ie all these things are allocated to pin 14

ellensp, 2021-12-05:
In Configuration_adv.h
you need to change 
```
#if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 3
```
to 
```
#if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 1
```

This is not a bug, this is a configuration error

closed 2021-12-05 by ellensp

Ruiaxe, 2021-12-05:
Many thanks, all works now.

Sorry for the false alarm, its my lack of experience, I'm learning.

Thanks for your time, once more.