MarlinFirmware/Marlin #23261: Stepper Motors stopped responding

uggels, 2021-12-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi im new to 3D printing and I was having an issue. I replaced the mainboard of my ender 5 plus with a BTT octopus with TMC2209 and I got my printer firmware configured and it seemed like everything was working properly until I tried to move the axis'. They motors moved in the opposite direction that I had wanted so, I went back in to the firmware and changed INVERT_X_DIR to true. I did this for the y and the z because they were all inverted. I flashed the firmware again but after I did this, the stepper motors stopped responding to manual commands. I'm not sure if they work if I execute a print through the printer. There seems to be no issues with the firmware after its done building but, for whatever reason, my stepper motors no longer respond. everything else seems to respond fine I just cant operate the stepper motors

### Bug Timeline

this is the first time that this has happened to me

### Expected behavior

Tried moving all axis on the tft display as well as on pronterface

### Actual behavior

nothing happened

### Steps to Reproduce

1. turn on machine
2. go to screen or connect to pronterface
3. input a movement command

### Version of Marlin Firmware

Marlin-bugfix-2.0x 

### Printer model

ender 5 plus

### Electronics

Bigtreetech Octopus V1.1

### Add-ons

BLTouch, BBT TFT50 V3.0, tmc2209

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7655031/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/7655032/Configuration_adv.zip)
[image]
[image]

borland1, 2021-12-05:
After uploading the firmware change, did you try initializing the EEPROM, either from the advanced configuration LCD menu or by sending the appropriate G-code command?  Might give it a try.

uggels, 2021-12-05:
> 

i tried doing that and the problem persits

borland1, 2021-12-05:
OK, For troubleshooting, other things to try:

-Disabling `SOFTWARE_DRIVER_ENABLE` in your `Configuration_adv.h` file.
-Disabling `SQUARE_WAVE_STEPPING` in your `Configuration_adv.h` file.
-Enable `MINIMUM_STEPPER_POST_DIR_DELAY` in Configuration_adv.h file with values greater than 20ns, such as:
`#define MINIMUM_STEPPER_POST_DIR_DELAY 200`

As a workaround, you could revert back to original firmware configuration and reversing the stepper motor directions in hardware by swapping pins on one connector side with each stepper motor cable.

ellensp, 2021-12-06:
Forgot to turn on 12/24v supply?

uggels, 2021-12-06:
Thank you for the help but I just ended up recompiling the firmware from scratch. Everything works now except for the BLTouch but I'm sure I can find tutorials for that. Thank you fornthe time I greatly appreciate it.

closed 2021-12-06 by uggels