MarlinFirmware/Marlin #23263: [BUG] BTT SKR Mini E3 V2.0 FAN HW PWM conflict with TONE timer on STM32 env

FyreDraygg, 2021-12-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I recently rebuilt my printer and finally decided to switch to the new STM32 env for my SKR Mini E3 V2. Everything worked quite well, except for the part cooling fan (FAN0) which shows erratic behaviour after any call to Buzzer::tone() occurs (thus including UI menu clicks) if FAST_PWM_FAN is enabled.

This appears to be a new form of the same issue with timer collisions previously seen with Maple:
on SKR Mini E3 V2 the controllable fans use pins PC6 and PC7, configured in the pinmap to their alternate function, meaning TIM3 (CH1 and CH2) is used to output hardware PWM to the fans when FAST_PWM_FAN is enabled in config.
However the Tone library function in the stm32 framework also uses TIM3 as specified in the variant.h file, ignoring any board build flags (which I suspect are incorrect anyway since TIMER_TONE should expand to a timer pointer, not a literal int?).

This breaks fan operation because in stm32 Tone expects to have exclusive control over the entire timer peripheral and when first called it creates a new HardwareTimer object which in turn both displaces any previous handle from the global HardwareTimer_Handle array and reinitializes the TIM peripheral, disrupting any channel configuration.

Back in Maple a similar conflict broke either fan or BLTouch/Servo operation depending on how timers were allocated to various functions, but I was able to mitigate the issue with a hacky workaround by saving the timer registers before the tone() call and restoring them afterwards. This enabled Tone to coexist peacefully with fan HW PWM since Tone would then only briefly alter the PWM signal to the fan.
Now with stm32 this pseudo-fix seems much more complicated and even more "hacky" since timers are managed through the HardwareTimer abstraction and it's not trivial to alter their settings without "acquiring ownership" over them by creating a new instance and resetting them in the process.

I'm by no means an expert of this codebase and all my understanding comes from carefully reading through the code, but as I see it either:
- A free TIMx is found to give to Tone exclusively (TIM1 and TIM2 are for thermal and step control, TIM3 is used for fan HW PWM, TIM5 I believe is used for Servo, TIM4 and TIM8 I don't know);
- A new speaker function is implemented within Buzzer independent from stm32 Tone so it can be made more "polite" to other users of the same timer;
- A sanity check is added to allow only FAST_PWM_FAN or SPEAKER, not both, to be enabled at the same time on this board/env (or whichever other conflicting combination of timer users).

Up until now I've managed to fix these small issues myself without bothering anyone by submitting reports or my hacky code, but since it's becoming an old issue which keeps resurfacing I think it's time to see it through.

### Bug Timeline

TIMx conflicts are an old issue with this board, both with the old Maple env and the new STM32 HAL. Tone is an especially infamous offender.

### Expected behavior

Fan operation should be independent from sounds played by the UI.

### Actual behavior

After any sound is played (UI click, M300, etc...) the fan behaves erratically, either off or 100% on (I think depending whether the output was high or low when the pin was disconnected from the timer peripheral by Tone reinitializing it)

### Steps to Reproduce

- Enable FAST_PWM_FAN, set FAST_PWM_FAN_FREQUENCY to one appropriate to your fan
- Enable SPEAKER
- Enable SOUND_MENU_ITEM
- Compile, upload, boot printer
- Disable sound in settings, save to EEPROM
- Reboot printer so that UI sounds are off
- Set a fan speed, the fan spins correctly
- Go to settings, enable UI sounds, the buzzer clicks in response
- The fan either stops or spins up to 100%.
- Go back to fan speed settings, set another speed
- The fan either changes speed or does nothing
- Push the button multiple times to produce clicks
- The fan either changes speed or does nothing

Checking with an oscilloscope, after a click the fan pin gets stuck either low, high or oscillates at around 1KHz.

### Version of Marlin Firmware

bugfix-2.0.x, updated to the latest commit at the time of writing

### Printer model

Creality Ender 3

### Electronics

BTT SKR Mini E3 V2.0 (512k)

### Add-ons

BLTouch, BTT TFT35 E3 screen, I don't think anything else uses timers

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

SD Card (headless)

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7656253/configs.zip)
(Lots of stuff enabled here because my chip has the magic extra 256k, let me know if I should upload a minimized version)

As a final note, I do realise that playing notes on the buzzer is not a critical feature and that several workarounds could be applied such as disabling SPEAKER, using SOFT_FAN_PWM, etc.

descipher, 2021-12-20:
[image]

Speaker is on TIM3
[image]
FAN is on TIM3 by default but has an alternate of TIM8
[image]

I believe this define in pins_BTT_SKR_MINI_E3__V2.0.h is out of date and would not work on this board as it would require -DTIMER_TONE=TIM3
```#define BOARD_CUSTOM_BUILD_FLAGS -DTONE_CHANNEL=4 -DTONE_TIMER=4 -DTIMER_TONE=4```

Have you tried using #define FAN_SOFT_PWM instead?

descipher, 2021-12-20:
To use the alternate pin mapping of TIM8 to PC7 as a PWM FAN output a variant definition and a Platform IO environment entry would be required as a final fix.

The current one is:
board_build.variant         = MARLIN_F103Rx

You can test this solution by altering the variant pin mapping file:
buildroot\share\PlatformIO\variants\MARLIN_F103Rx\PeripheralPins.c

Find line 222 and edit the PC_7 definitions to read as follows:

  // {PC_7,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_ENABLE, 2, 0)}, // TIM3_CH2
#if defined(STM32F103xE) || defined(STM32F103xG)
   {PC_7,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 2, 0)}, // TIM8_CH2
#endif

descipher, 2021-12-20:
Not certain but you may also need:
```#define BOARD_CUSTOM_BUILD_FLAGS -DTIMER_TONE=TIM8```

FyreDraygg, 2021-12-26:
I'm in the middle of a week-long series of urgent prints, so I will try out your kind suggestions as soon as it all finishes. Sorry for the delay, and thank you for your time.
In the meantime since my original post I had managed to make it work by replicating the hack I described with saving and restoring the timer state, however it involved editing not only the Buzzer logic but also adding some missing utility functions to the stm32 framework, which I feel is quite "dirty".
If TIM8 is not used by anything else, your solution of giving it to fan PWM would be so much cleaner.

Meanwhile, some things I can answer immediately:
- Yes, I tried `FAN_SOFT_PWM` and as far as I can tell it works properly. I can only test that with my oscilloscope however, since the fans I have don't like it (which is why I'm using hardware PWM in the first place)
- I think the `#define BOARD_CUSTOM_BUILD_FLAGS` has no effect. I have no idea what the order of includes and defines is, but I'm under the impression that code in the arduinostm32 framework ignores it entirely. The variant.h file must take priority over it, otherwise the `-DTIMER_TONE=4` (which I agree is outdated and meant for Maple) would cause a compile error and there's no code using `TIMER_TONE` outside the stm32 framework which would use the custom define. I remember the first part of the line (`#define BOARD_CUSTOM_BUILD_FLAGS -DTONE_CHANNEL=4 -DTONE_TIMER=4`) was added back in #21079 to solve a similar timer conflict in Maple, but when I tried it the change had no effect whatsoever (Maple kept using timer 8, which I confirmed with a simple print of the value over serial). I was about to report this finding back then, however the new STM32 HAL was going to replace Maple entirely and so I decided not to waste anyone's time on that and created the timer save/restore trick as an improvised fix.
- Correct me if I'm wrong, but I think Tone doesn't use the timer hardware outputs. It only uses timer interrupts and toggles in software whatever pin you tell it to use. If so, what pin the speaker is on doesn't matter and we could also move Tone to TIM8 (or whatever free TIMx) and keep fans on TIM3 if that's any cleaner

descipher, 2021-12-26:
Yes, the tone code only needs the timer for irq processing, so by freeing TIM3 it resolves the frequency conflict with the fan code. There are other ways to resolve it as well,  such as simply assigning a free timer to the tone function. For this board TIM8 may very well be the least likely to conflict again so thats why it’s a potential  the option here.

thinkyhead, 2022-01-05:
> …but when I tried it the change had no effect…

Well, `-DTIMER_TONE=4` is what would affect Maple, and `-DTONE_TIMER=4` is for STSTM32. It's easy to mix these up.

> Tone doesn't use the timer hardware outputs

Correct, a handler toggles the piezo.

FyreDraygg, 2022-03-08:
I apologise for my prolonged absence, I wanted to test this more extensively but the bot made me realise it's been far too long.

I changed the PeripheralPins.c file as suggested to:
```
  //{PC_6,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_ENABLE, 1, 0)}, // TIM3_CH1
#if defined(STM32F103xE) || defined(STM32F103xG)
  {PC_6,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 0)}, // TIM8_CH1
#endif
  //{PC_7,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_ENABLE, 2, 0)}, // TIM3_CH2
#if defined(STM32F103xE) || defined(STM32F103xG)
  {PC_7,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 2, 0)}, // TIM8_CH2
#endif
```
(PC6 controls the board cooling fan, which suffered from the same issue and thus needs to be changed likewise)

On my machine this works perfectly, and no issues have arisen in the time I've been using it with this change.
Should this pin mapping be changed to be the default for this variant? Can the simple change to PeripheralPins.c be merged upstream or would it affect other boards?
Ideally I would have tried to research this on my own and eventually send a PR, but as the bot reminded me, it's been too long already.

@thinkyhead on the topic of `-DTONE_TIMER` and `-DTIMER_TONE` flags: I'm well aware of the difference between them, sorry if I explained myself poorly. What I meant to say is that both flags do nothing even in their respective frameworks.
Changing either `-DTONE_TIMER` or `-DTIMER_TONE` does not affect STSTM32 just the same as changing them does not affect Maple.
(side note: correct me if I'm wrong but I think you switched them around in your reply, the Tone.cpp in STSTM32, line 112 uses `TIMER_TONE`. Maple used `TONE_TIMER` as far as I remember)
STSTM32 keeps using whatever is defined in variant.h and Maple kept using its own internal definition (I forget in which file).
I tested this by adding my own GCODE command to report timer assignments over serial and in both cases (Maple and STSTM32) it ignored the appropriate build flag.
Furthermore, if the `-DTIMER_TONE=4` flag actually affected STSTM32 it would generate a type mismatch error when compiling, since it can't use '4' as a TIM pointer which STSTM32 expects (check Tone.cpp line 112, same as above).
I don't know if this is something wrong with my specific build environment (I use VSCode with PIO) but at least for me it's been consistently behaving as if those build flag definitions are not included where they would then be used.
It's nothing major, but maybe it could still be useful to prevent headaches in the future.

As always, thank you for your time and patience

FyreDraygg, 2022-05-09:
I believe this is still relevant, please don't close it Mr. Bot

closed 2022-07-20 by github-actions[bot]