MarlinFirmware/Marlin #23265: [FR] Linear Advance Timing Control

kevinroosa, 2021-12-05:
### Is your feature request related to a problem? Please describe.

I have noticed on calibration test patterns and prints that the compensation provided by the Linear Advance feature "lags" a bit in its operation. The positive acceleration portion isn't _as_ affected, but there is still a noticeable amount of under-extrusion after the initial pressure ramp. The deceleration seems to be more affected, and creates a bump in width before the "pull" portion of the pressure adjustment.

### Describe the feature you want

I believe this could be solved with a timing adjustment feature, where additional parameters could be specified with M900 to "advance" pressure advance forward by a specified time.

### Additional context

Attached is an example. Machine used a TronXY X5SA Pro, Genuine E3D Titan extruder, ~800mm of Capricorn XS PTFE tubing and a Genuine E3D V6 hotend assembly. Print acceleration is at 2000mm/s and extruder jerk of 25mm/s, Hatchbox True White PLA @ 215C.

[image]

jmding8, 2021-12-17:
I've been wondering about this myself. I think the effect may be more visible in materials that are higher viscosity when molten. Although in my mind, it may be better to add a filament offset term that is a function of extruder acceleration, rather than a timing term.

XDA-Bam, 2021-12-17:
The way the Marlin planner currently is working, the idea of this FR can sadly not be implemented. The problem is, that the planner is always fully planning the very last block/movement added to the queue (apart from the backwards pass, which is limited in functionality) - it does not look ahead. It is essentialy "blind" and only handles the very block, which is being planned right now - with some limited info about the last block being passed on by select variables.

Therefore, the planner cannot determine, when the advance steps would have to be reduced - which is necessary for the feature you request. In the end, this limitation most strongly affects systems with significant hysterses, like Bowden setups. We need a completely new planning logic to gain the abiliy to fix/improve this.

jmding8, 2021-12-17:
when it handles the current block, is it able to determine what the current acceleration is?

XDA-Bam, 2021-12-17:
Yes, that is known. Have a look at the [planner](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/planner.cpp).

thinkyhead, 2022-01-05:
We are hoping to get some tutelage from @wavexx about Linear Advance so that we have parity with fixes that were long ago applied to Prusa Firmware.

thisiskeithb, 2022-08-04:
Linear advance should be improved after https://github.com/MarlinFirmware/Marlin/pull/24533

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

KondratokGK, 2024-07-07:
Same issue, almost 2 years ago.