MarlinFirmware/Marlin #23268: [FR] Setting to allow M206 offsets to add travel

ZetaPhoenix, 2021-12-06:
### Is your feature request related to a problem? Please describe.

I have my printer setup with an optical end stop that allows over traveling and I am attempting to use the M206 to store the exact position of the nozzle after homing but if I put a offset of 2 into the Z offset setting I am not able to move the machine to the print bed.

### Describe the feature you want

A enable flag that allows the software end stops to increase travel by the M206 offsets.

DerAndere1, 2021-12-10:
This is maybe the same feature I want. I would express it as follows: I need a way to set software endstop positions at runtime. Something like RRF command `M208` (M208 is taken for other purposes in Marlin, unfortunatelly) or something like GRBL commands `$131`, `$132` `$133`. Maybe we can repurpose M206. Attached image shows that M206 and G92 have the same effect, just different syntax. I feel like M206 is redundant because G92 is more flexible in combination with G53, G54,... coordinate systems. 

[image]

thinkyhead, 2022-01-05:
The `M206` command modifies the workspace, but the software endstops remain where they are. If you want to move lower in the Z dimension disable `MIN_SOFTWARE_ENDSTOP_Z` or `MIN_SOFTWARE_ENDSTOPS`.

ZetaPhoenix, 2022-01-05:
> The `M206` command modifies the workspace, but the software endstops remain where they are. If you want to move lower in the Z dimension disable `MIN_SOFTWARE_ENDSTOP_Z` or `MIN_SOFTWARE_ENDSTOPS`.

This is the issue I am running into. If I do not disable the SW end stops then I can not print the first few layers and if I disable the SW end stops I loose protection of crashing the print head into the bed.

I am asking for a setting to move the G54 workspace (print area WCS) to negative values based on the G53 (machine WCS) by the amount set in the M206 offsets so you can change out beds or tool heads and compensate the change in height if you limit switches support over travel.

DerAndere1, 2022-04-13:
Interestingly, I think the requested behaviour existed in Marlin from 2016. See commit message of https://github.com/MarlinFirmware/Marlin/commit/957e966d2d5fb1884adb1f982ad01d39d547eb36 .
To me it looks as if functionality of M206 was changed later in commit https://github.com/MarlinFirmware/Marlin/commit/8e5099fa0c3b35dea76941ff2de1fce08a6372c3 to make it update software endstops.